NVlabs/Deep_Object_Pose #1: Build problem--"pip install -r requirements.txt"

ZWYDeveloper, 2018-11-08:
hello!
I encountered a mistake while running this command "pip install -r requirements.txt",
Could not find a version that satisfies the requirement torch==0.4.0 (from -r requirements.txt (line 2)) (from versions: 0.1.2, 0.1.2.post1)
No matching distribution found for torch==0.4.0 (from -r requirements.txt (line 2))
so how I  solve it?

TontonTremblay, 2018-11-12:
We have not run into this problem while testing, can you tell me more about your pip and python version as well as the system you are working with. I will try to reproduce it. 

Meanwhile one way around it would be to remove `==0.4.0` in `requirements.txt`. Or an other way would be to install pytorch manually and remove the line `torch==0.4.0` from the `requirements.txt` file. You can install pytorch manually following these steps: https://pytorch.org/ - about half way through the page. I hope this helps.

ZWYDeveloper, 2018-11-12:
Thank you for your reply, I  use python2.7 in ubuntu16.04LTS ,  and use the PIP is also the latest version. 
I'll try it your way, thank you very much

ZWYDeveloper, 2018-11-12:
> We have not run into this problem while testing, can you tell me more about your pip and python version as well as the system you are working with. I will try to reproduce it.
>
> Meanwhile one way [...]

hi!
I have tried it your way ,but also some  error
Screenshot from 2018-11-12 10-10-56
I feel that there is something wrong with the python I used.

ZWYDeveloper, 2018-11-12:
[image]

TontonTremblay, 2018-11-13:
I would suggest to try using the provided docker image, it might be a little simple to get the requirements going.

ZWYDeveloper, 2018-11-14:
> I would suggest to try using the provided docker image, it might be a little simple to get the requirements going.

I reinstalled the Ubuntu system, the problem has been solved, the compilation is very smooth, so if I want to achieve the effect of dope in my camera, what should I do? Thank you very much for your reply.

ZWYDeveloper, 2018-11-14:
I compiled it smoothly. I didn't see any images according to the official steps. I tried to open the USB camera in ROS and run the dope. py file, but I still couldn't see the effect.

TontonTremblay, 2018-11-15:
In rviz can you see what the camera is seeing? I.e., can you subscribe to the camera node?

ZWYDeveloper, 2018-11-16:
Yes, I can see my camera node in Rviz, but the test image doesn't seem to get the desired effect, so can you provide me with a picture you trained to use? Thank you