NVlabs/Deep_Object_Pose #2: Can't get the image

smarthhy, 2018-11-09:
when i run rosrun dope camera.py and rosrun dope dope.py ，
@hhy-System-Product-Name:~/own-project/catkin_ws$ rosrun dope dope.py 
Loading DOPE parameters from '/home/hhy/own-project/catkin_ws/src/Deep_Object_Pose/config/config_pose.yaml'...
    Parameters loaded.
Loading DOPE model '/home/hhy/own-project/catkin_ws/src/Deep_Object_Pose/weights/soup_60.pth'...
    Model loaded in 5.50860905647 seconds.
Running DOPE...  (Listening to camera topic: '/dope/webcam_rgb_raw')
Ctrl-C to stop

But i can't get the image.Can you help me?thanks!

ZWYDeveloper, 2018-11-09:
hello,how do you build the dope?when i run the command of  "pip install -r requirements.txt"encountered any problem,
[image]
so how do you solved it?

smarthhy, 2018-11-10:
@ZWYDeveloper sorry，I have not encountered this problem, the installation is very smooth, I use the version of python is 2.7 ，Do you use jetson?

ZWYDeveloper, 2018-11-12:
> @ZWYDeveloper sorry，I have not encountered this problem, the installation is very smooth, I use the version of python is 2.7 ，Do you use jetson?

Thank you for your reply, I also use python2.7, I follow the steps of the official document, and use the PIP is also the latest version. I don't know why some packages in the requestment.txt file cannot be installed, but I am trying to solve it

TontonTremblay, 2018-11-12:
As for listening to the image problem, you need to deploy a camera node. We provide a very crude one with the package: 
```
$ rosrun dope camera.py  # Publishes RGB images to `/dope/webcam_rgb_raw`
```
You need to run this ros node in a new terminal which has ros and the devel sourced. If you want to use the poses make sure your camera is calibrated correctly otherwise the projections wont look correct.

dj4park, 2019-07-18:
Does this package support installation on a Jetson TX2 board?

TontonTremblay, 2019-07-22:
It should work on Jetson, I have tested it last year and I was getting around 1-2 frames per second.