NVlabs/Deep_Object_Pose #5: Training Code

aditya2592, 2018-11-20:
Hi,

Will the training code for DOPE be made publicly available ?

TontonTremblay, 2018-11-21:
In the short term I don't think so. Why would you need it?

aditya2592, 2018-11-21:
Hi @TontonTremblay. This is for a manipulation project at CMU. We wanted to train it for other objects in the YCB dataset and test it with our robot.

TontonTremblay, 2018-12-20:
We made the training code available if you are interested, please check: https://github.com/NVlabs/Deep_Object_Pose/blob/master/src/training/train.py

MohamadJaber1, 2019-04-04:
@aditya2592 as he mentioned, I am highly interested in implementing this great work, but I want it from my own object, is there a possiblity to explain it?

eyildiz-ugoe, 2019-05-22:
I guess I am just another one out there who really wants to try this with other objects. Any updates on this?

mintar, 2019-05-22:
Short answer:

1. Generate a mesh of your object. Either use a CAD model if available, or use e.g. [Meshroom](https://alicevision.github.io/#meshroom).
2. Generate a synthetic dataset of your object using [NDSS](https://github.com/NVIDIA/Dataset_Synthesizer).
3. Train the network using train.py.

eyildiz-ugoe, 2019-05-22:
@mintar  Thank you for the explanation, is there a walkthrough on the second point? I've checked the Github page but couldn't find "how to train with your own model" kind of thing.  There is information only about how to visualize the existing dataset, or have I missed something?

mintar, 2019-05-22:
https://github.com/NVIDIA/Dataset_Synthesizer/blob/master/Documentation/NDDS.pdf

mintar, 2019-05-22:
Also check out the videos: https://github.com/NVIDIA/Dataset_Synthesizer/tree/master/Videos

liangzhicong456, 2019-05-22:
@mintar Thank you for summing up. I want to konw how to randomly change the pose, and also import the real world background in the NDDS. Now I just randomly modify the values of RX, RY,  but not satisfied. Another point I want to know is that my model size must strictly follow the true size? I hope you can get your patient answer, thanks!

mintar, 2019-05-24:
The model size must be exactly the same as the true object size. Your other question is answered in #45.

liangzhicong456, 2019-05-24:
@mintar Yes, thanks for your patient reply!

aditya2592, 2019-06-23:
@mintar I am trying import the aligned YCB object models (downloaded using nvdu_ycb) into NDDS and generate scenes with random poses. However importing the .obj file doesn't load the color information in texture_map.png. How do we import that?

zhezhou1993, 2019-06-24:
You can use Blender to import your .obj with texture then save as .fbx file which can be imported into unreal.

aditya2592, 2019-06-24:
I was able to directly import the .obj. However it only worked in a fresh unreal project. I think the NDDS project was too big and causing the import to freeze when importing with material information. I then copied the assets from the fresh project to the NDDS project

closed 2019-08-23 by aditya2592