NVlabs/Deep_Object_Pose #6: Unable to open camera for capture.  Is camera plugged in?

qq4060, 2018-11-22:
~/catkin_ws$ rosrun dope camera.py
ERROR:  Unable to open camera for capture.  Is camera plugged in?

TontonTremblay, 2019-01-06:
It would be straight forward to make it work on an images coming from rosbag for example.

Luedeke, 2019-01-17:
You can use this small rosbag: https://github.com/Luedeke/Deep_Object_Pose/blob/master/rosbag_images_sorted.bag
with this two files:
https://github.com/Luedeke/Deep_Object_Pose/blob/master/src/dope_rosbag.py
https://github.com/Luedeke/Deep_Object_Pose/blob/master/config/config_pose_rosbag.yaml

AOL283, 2020-04-02:
Hi,
Could you please provide better clues on how to make it work with images from a bag file.
What needs to be change? Which files? I guess the pipeline is the same (roscore, script1, script2)..
but DOPE doesn't support the "play" argument instead of "camera"
In the files above, they even modified the location of the weights, and is therefor confusing for someone new to the field.
Thank you,

TontonTremblay, 2020-04-02:
I never tried. Have you tried changing the topics in DOPE to listen to your rosbag playing?

avinashsen707, 2020-04-16:
> ~/catkin_ws$ rosrun dope camera.py
> ERROR:  Unable to open camera for capture.  Is camera plugged in?
> 

You can write a python program in seperate for capturing.
There are alot of samples in open source for capturing... provided u need to install the drivers and sdk of realsense.

closed 2025-08-08 by nv-jeff