NVlabs/Deep_Object_Pose #7: cannot get any image from camera

qq4060, 2018-11-23:
hi
I just run and cannot get any image from camera ,ut shows:
```
~/catkin_ws$ rosrun dope camera.py 
Publishing images from camera 0 to topic '/dope/webcam_rgb_raw'...
Ctrl-C to stop
^Cqq4060@qq4060-To-be-filled-by-O-E-M:~/catkin_ws$ rosrun dope camera.py 
Publishing images from camera 0 to topic '/dope/webcam_rgb_raw'...
Ctrl-C to stop
```
 
and also shows 
```
~/catkin_ws$ rosrun dope dope.py
Loading DOPE parameters from '/home/qq4060/catkin_ws/src/dope/config/config_pose.yaml'...
    Parameters loaded.
Loading DOPE model '/home/qq4060/catkin_ws/src/dope/weights/soup_60.pth'...
    Model loaded in 4.77608513832 seconds.
Running DOPE...  (Listening to camera topic: '/dope/webcam_rgb_raw')
Ctrl-C to stop

```
Thank you very much!

TontonTremblay, 2018-11-23:
This looks great to me. Do you see the `/dope/webcam_rgb_raw` topic in rviz?

qq4060, 2018-11-23:
OH I CANNOT SEE ANY OTHER THINGS  OPEN OR RVIZ

ZWYDeveloper, 2018-11-30:
Hi, have you tried to train your model with pytorch?

qq4060, 2018-11-30:
> Hi, have you tried to train your model with pytorch?

SORRY   HOW TO TRAIN?
 use this  line"$ rosrun dope dope.py [my_config.yaml]  # Config file is optional; default is `config_pose.yaml`"?

ZWYDeveloper, 2018-11-30:
First you may need the six models or pictures provided by invidia. In rviz, set the image topic to: / dope / rgb_points, and then point the camera at your model or picture.

qq4060, 2018-11-30:
Thanks, Is the six models or pictures provided by invidia in this step ?
YCB 3D Models
DOPE returns the poses of the objects in the camera coordinate frame. DOPE uses the aligned YCB models, which can be obtained using NVDU (see the nvdu_ycb command).

ZWYDeveloper, 2018-11-30:
Yes, but you can search the pictures of the six models online first. Nvdu and NDDs are used to train YCB model data. It seems that pytorch will be used in the end. It's a bit troublesome. We can exchange and study together at that time. I hope it will be helpful to you.

qq4060, 2018-11-30:
how about this they said no train code

 https://github.com/NVlabs/Deep_Object_Pose/issues/5

ZWYDeveloper, 2018-11-30:
No, you can study it. I'm also trying to train my model.

qq4060, 2018-11-30:
OK  Thanks a lot!

NULL-li, 2022-02-26:
您好，我的rviz添加相机后没有图像，请问你解决了吗？如果方便的话可以跟您聊聊吗？我的微信y15537874728

closed 2025-08-08 by nv-jeff