NVlabs/Deep_Object_Pose #8: Training on Custom data & CPU machine

nikhiljaiswal, 2018-12-05:
Hi,
How can we train the network on our custom data?
Is GPU required during test time also?

TontonTremblay, 2018-12-20:
I will refer you to the newly uploaded training script: https://github.com/NVlabs/Deep_Object_Pose/blob/master/src/training/train.py.

SHZ2006, 2019-09-05:
Hi,
Have you taken the pictures in UE by handling the camera by hand? Or there is a code to do it automatically?
I mean when I put the object in UE and start the snapshots, just the environment changes and the object and camera are fixed. Shall I move the camera by hand for each picture manually?
Best,

TontonTremblay, 2019-09-08:
https://github.com/NVIDIA/Dataset_Synthesizer 
I will refer you to the NDDS tool for generating synthetic data. There is a component called the orbital camera, it will move the camera for you :P

avinashsen707, 2019-12-18:
> I will refer you to the newly uploaded training script: https://github.com/NVlabs/Deep_Object_Pose/blob/master/src/training/train.py.

is link looks like error 404

mintar, 2019-12-18:
The script location has changed, it's now here:

https://github.com/NVlabs/Deep_Object_Pose/blob/1ee8e0ddaa01886785ba176581582947008440d4/scripts/train.py

avinashsen707, 2019-12-18:
I am a Post Graduate student of Nodal Centre for Robotics and AI, Goverment Engineering College Thrissur, Kerala, India.
I am currently doing a Master Project on Bin Picking of 3D objects, using an Aubo collaborative Robot.
Currently i am trying Nvidia-DOPE library for Object Detection and Pose Esimation for custom domains. I make use of UE4 Engine for data set generation.However, I am not able to install required plugins from running NDDS project file.

i am working in these platforms:
OS : Ubuntu 16.04 LTS
GPU : Nvidia Quadro P4000 (8GB)
Driver : 384.130
Cuda : 10.1
UE4 : 4.22 release

As you have already implemented custom dataset using UE4 engine, I request you to share  your apprach  in this reagard!

@TontonTremblay 
@mintar

TontonTremblay, 2019-12-18:
Are you having issues with NDDS or with dope itself. If you are having issues with NDDS, please make an issue on the NDDS github please.

avinashsen707, 2019-12-21:
@TontonTremblay let me look into it

closed 2025-08-08 by nv-jeff