NVlabs/Deep_Object_Pose #11: Training dataset

QuanNguyen94, 2018-12-24:
hi, I have implemented completely my training code, I want to test my code 's performance on your training dataset to have obvious comparing, can you send to me your training dataset of 1 object, that 's enough.

Abduoit, 2018-12-24:
can I know please how did you train for new object ?

TontonTremblay, 2018-12-24:
Natively it will work with this dataset: https://research.nvidia.com/publication/2018-06_Falling-Things and with data generated with https://github.com/NVIDIA/Dataset_Synthesizer.

SHZ2006, 2019-08-26:
Hi, I am trying to train another object from your dataset:  https://research.nvidia.com/publication/2018-06_Falling-Things 
I have put banana to train and it get 5 epoch after 4 days. I want to know is it normal or it should be faster?
Thanks
My spec:
Intel® Core™ i7-8700 CPU @ 3.20GHz × 12
GeForce GTX 1060 6GB/PCIe/SSE2
Ubunta 18
Memory 32 GB

SHZ2006, 2019-08-28:
Thank you for your answer.
I want to know when I train a new object, should I just add it in dope/config/config_pose.yaml   as same as others and put the trained weights in weights folder? Thats it?

Abdul-Mukit, 2019-08-29:
Yes, that should be it.

aditya2592, 2019-08-31:
While training for a single object, does the training script exclude images not containing that object in the FAT folder?

SHZ2006, 2019-09-17:
Hi, when I want to continue training, I use for example --net train_011_banana_16k and I get error and if I use --net train_011_banana_16k/net_epoch_11.pth   it starts from epoch 1. How can I continue training from epoch 12?
When I want to continue training, should I disable pretrained ImageNet?
Thanks in advance.

TontonTremblay, 2019-10-08:
@aditya2592 not it does not exclude images that do not contain the object of interest. This is needed in order to avoid false positive. 

@SHZ2006 The script does continue training from the weights you gave it, the counting just reset. It would not be hard to change the epoch value in the script to get the behaviour you want. You do not need to disable --pretrained. The script first loads the base model, then it loads the save weights and you are ready to go.

BigRays, 2020-11-03:
> 嗨，我已经完全实现了我的训练代码，我想在您的训练数据集上测试我的代码的性能，以进行明显的比较，是否可以将1个对象的训练数据集发送给我，就足够了。

训练数据集时，FAT训练集文件怎么放置的？