NVlabs/Deep_Object_Pose #12: Training for new object ??

Abduoit, 2018-12-24:
Thanks for your effort 

I want to try training with new object. I tried to use file [train.py](https://github.com/NVlabs/Deep_Object_Pose/blob/master/src/training/train.py) with fat dataset. I am not sure how to start training 
any help will be highly appreciated 

```
abdu@grasp-uoit:~/catkin_ws/src/dope/src/training$ python train.py --data /home/abdu/fat --object soup --outf soup --gpuids 0 1 2 3 4 5 6 7 
/usr/local/lib/python2.7/dist-packages/requests-2.19.1-py2.7.egg/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
start: 12:56:49.259845
load data
Traceback (most recent call last):
  File "train.py", line 1261, in <module>
    pin_memory = True
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/utils/data/dataloader.py", line 802, in __init__
    sampler = RandomSampler(dataset)
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/utils/data/sampler.py", line 64, in __init__
    "value, but got num_samples={}".format(self.num_samples))
ValueError: num_samples should be a positive integeral value, but got num_samples=0
```

yuzw213, 2018-12-25:
hi， 
I think it might be because the object name behind the '-- object' is wrong. It doesn't exist in the fat dataset. You can look at the structure of the fat dataset. I tried running it directly: Python train.py can see the results. The fat data set is too big. I'm still trying to download it.

Abduoit, 2018-12-25:
Thanks @TontonTremblay and @yuzw213 

I changed `png` to `jpg` as stated in `FAT` dataset. 

The issue now is not a bug but the GPU is running of out memory. I am using single GPU 1080 with 8 GB memory. I am not sure if it is possible to continue training by reducing the data dimension. ???
Any suggestion please ??

```
abdu@grasp-uoit:~/catkin_ws/src/dope/src/training$ python train.py --data /home/abdu/fat --object scissors --outf scissors --gpuids 0
/usr/local/lib/python2.7/dist-packages/requests-2.19.1-py2.7.egg/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
start: 14:28:24.447948
load data
training data: 7688 batches
load models
Traceback (most recent call last):
  File "train.py", line 1385, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1333, in _runnetwork
    output_belief, output_affinities = net(data)
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/nn/parallel/data_parallel.py", line 141, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "train.py", line 160, in forward
    out1_2 = self.m1_2(out1)
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/nn/modules/container.py", line 92, in forward
    input = module(input)
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/abdu/.local/lib/python2.7/site-packages/torch/nn/modules/conv.py", line 320, in forward
    self.padding, self.dilation, self.groups)
RuntimeError: CUDA out of memory. Tried to allocate 39.12 MiB (GPU 0; 7.93 GiB total capacity; 7.15 GiB already allocated; 7.75 MiB free; 1.35 MiB cached)
```

Abduoit, 2019-01-05:
Now I tried to decrease the `batchsize` to `16 `. And I also changed `loss.data[0]` to `loss.item()`. And inside the FAT folder, there is only the object that I want to train for, I removed the others. After training has finished and I got the file `net_epoch_60.pth`, I tried to test it but nothing detecting.

Do you think the object name is not correct ??

`python train.py --data /home/abdu/fat --object scissors --outf scissors --gpuids 0`

it should be 

`python train.py --data /home/abdu/fat --object 037_scissors_16k --outf 037_scissors_16k --gpuids 0`

Should I also remove the folder `mixed ` inside FAT structure to make it faster?

TontonTremblay, 2019-01-06:
How much data are you using? You might want to change the learning rate as well. Only training on FAT for some objects might not be enough. The scissor is quite a small object. When I was developing DOPE I had create debug tools to look at the output of the neural network. So what I would do first would be to look at the output of the neural network on the training images to see that it actually outputs the belief maps. Then I would test on other images.

Abduoit, 2019-01-06:
I used the whole data inside `fat/single/037_scissors_16k`. What should I make the learning rate what do you suggest ??
it is like this rightnow
```
parser.add_argument('--lr', 
    type=float, 
    default=0.0001, 
    help='learning rate, default=0.001')
```

saratrajput, 2019-03-19:
@Abduoit Hello! Were you able to solve the problem with CUDA memory? I'm running into the same issue and I've tried changing batch size and learning rate.

Abduoit, 2019-03-19:
@saratrajput did you try to use only one folder (`single`) in the dataset (`fat`). I removed the folder `mixed` and I have inside `single` only one object folder which is the one that I want to train for.

saratrajput, 2019-03-20:
@Abduoit Sorry, it was a related but not the same problem. I solved it by adding --shm-size 8G to the docker run command. And also earlier, one of the gpus was being used. My bad. By the way, I still have the mixed folder inside fat. I don't think it makes a difference. Thanks for the quick reply.

Abduoit, 2019-03-20:
@saratrajput when you have the mixed folder inside fat, the training process ll be slower since you are uploading more data.

saratrajput, 2019-03-21:
@Abduoit Oh, okay. I understand. Thanks!

Dutzel, 2019-04-12:
@saratrajput &  @Abduoit now I am the third in row facing the CUDA memory problem... I've also tried changing batch size and learning rate but it didnt fix my problem. I am trying to train on my own little dataset, so there is no structure like in the fat dataset. Furthermore I am not using docker so I am not able to run the docker run command. 

Any other ideas that may help me? thank you :)

saratrajput, 2019-04-15:
@Dutzel At times this CUDA memory problem can also be cause because of incompatibility between the GPU and the pytorch version. Maybe try on a different GPU or give it a try with docker first.

dj4park, 2019-07-24:
Hello, I am trying to train DOPE with an object (a plastic water bottle) that is not part of the FAT dataset. Is it possible for me to generate a custom FAT dataset using [NDDS](https://github.com/NVIDIA/Dataset_Synthesizer)? If so, how would it be done?

I am also restricted to working with a NVIDIA Quadro P3200 Mobile and was wondering if Google Colab could be used to train the data.

As an undergrad student working for a research lab, I am pretty inexperienced in the field. Any help or suggestions would be greatly appreciated. Thank you!