NVlabs/Deep_Object_Pose #13: Train yours own dataset?

yuzw213, 2018-12-26:
Hi, if I want to train one of my own objects, but because of the network here, the fat dataset cannot be downloaded. I want to make a dataset myself. Does my dataset structure have to be the same as the fat dataset? Can I don't need the "mixed"directory ? What are the requirements for the size of the object image generated by NDDS? I hope to hear from you, thank you very much!

BIGJUN777, 2019-01-01:
Hello, have you got any idea about training your own dataset? I also want to train the objects of interest, but don't know how to do. It seems not easy to get the labeled dataset.

BIGJUN777, 2019-01-02:
Well, please keep in touch here. I gonna work on NDDS too.

Abduoit, 2019-01-03:
@yuzw213 
I tried to download the FAT dateset first, then after finish downloading, I give the path to the datatset, I haven't done training yet because I got issue related to the CUDA memory and I don't know how to get around it. You can check the issue [HERE](https://github.com/NVlabs/Deep_Object_Pose/issues/12#issuecomment-449873541)

BIGJUN777, 2019-01-04:
@Abduoit Did you try to train a small subset? I didn't download the FAT Dateset for the bad internet speed here. Maybe you can share a small part of this subset. Thank you.

yuzw213, 2019-01-04:
@Abduoit  By reducing the value of batchsize, you can try setting it to 16 or 8, as well as the fat's "single" directory, where you can just put a data on the object you want to train

Abduoit, 2019-01-04:
thanks @yuzw213 
I start training for scissor object, until `Train Epoch: 15` `LOSS: 0.000344149710145`. I got file `net_epoch_15.pth`. But when I tried testing, it doesn't detect anything. I wonder if the LOSS not enough and I should leave it for more training time.

Abduoit, 2019-01-04:
@BIGJUN777 do you want I send you dataset for one object only.

BIGJUN777, 2019-01-05:
> @BIGJUN777 do you want I send you dataset for one object only.
Yeah, Thank you. My Email: `zhijunliang777@gmail.com` or `1569240491@qq.com`.

Abduoit, 2019-01-05:
> > @BIGJUN777 do you want I send you dataset for one object only.
> 
> Yeah, Thank you. My Email: `zhijunliang777@gmail.com` or `1569240491@qq.com`.

@BIGJUN777 which object do you want ?

Abduoit, 2019-01-05:
@yuzw213  Could you please tell me if the object name should be exactly the same like 

`python train.py --data /home/abdu/fat --object scissors --outf scissors --gpuids 0
`
it should be

`python train.py --data /home/abdu/fat --object 037_scissors_16k --outf 037_scissors_16k --gpuids 0 `

And should I also have folder `mixed `inside FAT dataset

BIGJUN777, 2019-01-06:
@Abduoit Any object is ok. I just want to have a look at training. Thank you.

yuzw213, 2019-01-06:
@Abduoit I think the name can be picked up freely, but it has no effect on training. The key is the "mixed" directory. I am not sure if it is necessary to train a single object.

Abduoit, 2019-01-06:
@yuzw213 
I tried to trained the object `scissors` but nothing detected. I don't know exactly where I am mistaken. Maybe learning rate need to be changed or something else. How about you, did you try to train your own object. ??

yuzw213, 2019-01-07:
@Abduoit  I made a mistake. The name after "--object" should be the same as the class name of the object you want to train. So I suggest you try again with the following command. In addition, the "mixed" directory is also needed.
`python train.py --data /home/abdu/fat --object 037_scissors_16k --outf 037_scissors_16k --gpuids 0`

Abduoit, 2019-01-07:
@yuzw213 
I already tried as you suggested, included "mixed" folder. The training process was very slow, I left it over a night. But I guess it also does't work, nothing detected. 
I will try to train it again with another object which is bigger like "drill", because "scissors" is much difficult and smaller.

TontonTremblay, 2019-01-07:
The original DOPE weights trained on 4 or 8 P100 or V100 for more than 24h with batchsize of 128 0r 224 on 220k images total. I would be surprised you would accomplish similar results on single GPU overnight. DOPE is quite a large network and needs quite of bit of time to train. Please refer to this example (https://github.com/NVlabs/Deep_Object_Pose/issues/4) for visualizing the raw output of the network which will help you determine the quality of the training.

Abduoit, 2019-01-07:
Thanks @TontonTremblay 
I checked the example, but I am not sure how to visualize the raw output, or to check the belief maps.

TontonTremblay, 2019-01-07:
You will have to write that code yourself sorry.

Abduoit, 2019-01-08:
Thanks for your reply @TontonTremblay 
Do you think there's an alternative way to train for one object, I mean by leasing DGX Nvidia station or any online training service.  ??

TontonTremblay, 2019-01-15:
I have forgot a loss in the training code which I updated today, sorry about this. Please update the code to include these changes if you want to train.

BIGJUN777, 2019-01-15:
@TontonTremblay 
Thank you.

Actually, I updated the code as you did a day ago and tried training again. It got a better result but is not enough. How can I get or create the DR images dataset?

yuzw213, 2019-01-15:
@TontonTremblay  Thank you for your update. I will update the code and try again.

@BIGJUN777  Randomized image data can be generated by NDDS, but the amount of 220K data is a bit too much for me, but I think increasing the amount of data as much as possible should make the results better

TontonTremblay, 2019-01-15:
@BIGJUN777 I am glad you were able to make the changes accordingly, again sorry about this. @yuzw213 is correct you need to use NDDS to generate the DR data. I have seen good results with 50k images mixing 50% of each. Pose estimation needs a lot datat to cover all possible poses in large variety.

Luedeke, 2019-02-15:
I am also currently working on my own training data set. I have a problem with one of my models. The Static Mesh looks like it should. But if I add this mesh to the NVAnnotetedActor it gets dirty or very pixelated see appendix. Can someone explain my mistake to me?

Static Mesh;
[image]

NVAnnotetedActor:
[image]
[image]

paul-shuvo, 2019-02-19:
> @Abduoit I made a mistake. The name after "--object" should be the same as the class name of the object you want to train. So I suggest you try again with the following command. In addition, the [...]

@yuzw213 , I tried training the `banana` object today without the `mixed` folder and it was still able to run thorugh the 60 epochs albeit the model didn't detect the `banana`. So, I have a few questions-
1. Would it still train if the `mixed` folder is not present?
2. If it does, is the `mixed` folder very important for detection?
3. I have cloned the repository on 6th Jan and 15th Feb, while for the earlier one the follwoing command works `python train.py --data /home/fat --object banana --outf banana --gpuids 0` but for the latter one when I ran the same command it throws an error like it's expecting num_sample to be a positive integer value, so should I do `python train.py --data /home/fat --object 011_banana_16k --outf 011_banana_16k --gpuids 0` instead? Just to be sure, I tried with the cracker_box, but the model didn't detect anything. Thus, I believe I'm making some mistakes on training it. What is the proper way of training objects from the FAT dataset?

yuzw213, 2019-02-19:
@paul-shuvo 
Regarding the "mixed" directory, if you just want to train an object, the "mixed" directory can not be needed. I think if you want to detect multiple objects at the same time, you may use it. If you're training with the "fat" data set, you should use the object class name "011 _ banana _ 16k"

paul-shuvo, 2019-02-19:
@yuzw213 I tried with 

`python train.py --data ~/fat --object 003_cracker_box_16k --outf 003_cracker_box_16k`

But it gives me the error-

```
start: 11:26:43.686520
load data
Traceback (most recent call last):
  File "train.py", line 1261, in <module>
    pin_memory = True
  File "/home/lab-admin/anaconda3/envs/py36/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 802, in __init__
    sampler = RandomSampler(dataset)
  File "/home/lab-admin/anaconda3/envs/py36/lib/python3.6/site-packages/torch/utils/data/sampler.py", line 64, in __init__
    "value, but got num_samples={}".format(self.num_samples))
ValueError: num_samples should be a positive integeral value, but got num_samples=0

```
Do you have any idea on what might be causing this? I'm using python 3.6.

Abduoit, 2019-02-19:
@paul-shuvo 
try to use the latest file `train.py`, the command should like this 

`python train.py --data /home/fat --object 003_cracker_box_16k --outf 003_cracker_box_16k --gpuids 0`

paul-shuvo, 2019-02-19:
@Abduoit 
i trained it for 138 epoch, but the model doesn't detect anything, my fat folder structure is-

>~\fat
>>\single
>>>\003_cracker_box_16k

Also, how much time did it take for you to train 138 epochs? I'm using 1080 and it took around 5 minutes.

Abduoit, 2019-02-19:
it will definitely take long time with single GPU (may be 30 hours), I also removed other objects folders from folder-single. So the structure is like fat--->single---->only power drill folder.

paul-shuvo, 2019-02-19:
@Abduoit 
When I'm trying to train in therminal it shows `training data: 0 batches`, is it normal?
[image]

Abduoit, 2019-02-19:
what is the --batch_size in train.py , and what is the --epochs ??

paul-shuvo, 2019-02-20:
I found this 
```python
parser.add_argument('--batchsize', 
    type=int, 
    default=32, 
    help='input batch size')
```
I think the batch size should be 32; I have no clue what I am doing wrong!

TontonTremblay, 2019-02-20:
The problem is due that you are not loading images due to me hard coding loading pngs. Thus ctrl-f 'png' and change it for jpg. Let me know if this helps. I will push an extension to manually change it with a command line.

Luedeke, 2019-03-05:
Hello,
could someone tell me how to randomize the camera in NDDS?
I don't get it.

Thank you

yuzw213, 2019-03-08:
hi，You can try this component "OrbitalMovement". My own model is still not very good at identifying real objects. I think this may be related to the fidelity of the object model, scene lighting, and the amount of data trained.

Luedeke, 2019-03-14:
Hello @yuzw213 ,
thanks four your answer. 
sorry I think I'm blind, where can I find "OrbitalMovement"?

And do you have any idea if it is possible to train Dope without stereo camera recording? In NDDS neither "SceneCapturer_Simple" nor "SceneCapturer_IntelRealSense_SR300" seems to produce stereo images like in the dataset of DOPE. So it doesn't create files with "Number.left.png" and "Number.right.png" (StereoCamera picture) it just creates "Number.png" (you can see in NDDSCapturer). 

[image]
[image]

TontonTremblay, 2019-03-14:
You do not need to have stereo camera, we created FAT dataset with the capacity to investigate stereo algorithms. Dope is built for single rgb camera only, so yes you can use the SceneCapturer_Simple.

Luedeke, 2019-03-14:
@TontonTremblay Oh good phew, thank you very much!

chibai, 2019-03-20:
> @BIGJUN777 I sent you the data for one object, please check your email

Would you mind send me one object to train as well??
My email is 470598003@qq.com， thanks a lot!

BIGJUN777, 2019-03-22:
> > @BIGJUN777 I sent you the data for one object, please check your email
> 
> Would you mind send me one object to train as well??
> My email is [470598003@qq.com](mailto:470598003@qq.com)， thanks a lot!

I have sent you the dataset. Please check the email.

chibai, 2019-03-22:
> > > @BIGJUN777 I sent you the data for one object, please check your email
> > 
> > 
> > Would you mind send me one object to train as well??
> > My email is [470598003@qq.com](mailto:470598003@qq.com)， thanks a lot!
> 
> I have sent you the dataset. Please check the email.

Thanks a lot, I've download the cracker object

Luedeke, 2019-04-02:
Hello, @TontonTremblay ,

Do you have an overview picture of DOPE? 
Your paper doesn't contain one or does someone else have an overview of DOPE?

Thank you very much.

TontonTremblay, 2019-04-02:
Something like this? https://drive.google.com/open?id=1BwPceISmXZuupwjoGxfp9W37Dv6I2QLn

Luedeke, 2019-04-02:
@TontonTremblay  Thank you very much for your answer.
I meant an architecture or model overview of DOPE.
I found an rough architecture overview in your link, thank you very much :).

TontonTremblay, 2019-04-02:
That is the best I have, you can always look at the code for the details of
the model, I am happy to answer any questions you have about it.

Dutzel, 2019-04-11:
Hey guys, I tried to follow your instructions to get into training my own data later... So I also started with the "fat"-dataset and used the  command "python train.py --data /home/dustin/fat --object 003_cracker_box_16k --outf 003_cracker_box_16k --gpuids 0" which leads me to the same error mentioned in this issue "ValueError: num_samples should be a positive integeral value, but got num_samples=0"

What did I do wrong? Was there a change in the structure of the fat dataset? For me its
e.g.
- /fat/mixed/kitchen_0/data
- /fat/mixed/kitchen_1/data
...
and
- /fat/single/002_master_chef_can_16k/kitchen_0/data
- /fat/single/003_cracker_box_16k/kitchen_0/data

saratrajput, 2019-04-12:
@Dutzel  You might be using **jpeg** images. The code is looking for **png** images. Just change the extension in code or download the png dataset.

Dutzel, 2019-04-12:
@saratrajput you are right... i assumed the data set would already be optimally tailored to the project... sorry for that :)

BIGJUN777, 2019-04-14:
@Luedeke 
> Hello, @TontonTremblay ,
> 
> Do you have an overview picture of DOPE?
> Your paper doesn't contain one or does someone else have an overview of DOPE?
> 
> Thank you very much.

Hi, Luedeke,

Here is a PPT I created to introduce DOPE in my group a few months ago. Feel free to check it [Here](https://drive.google.com/file/d/0BzCuJEb-f0YLeGt5M203RlNhbGVUbzEzZWx4RmZKenZNR0VF/view?usp=sharing). 
It is also recommended that you look at the `train.py` file (the **NEURAL NETWORK MODEL** part) to know how to bulid  this neural network.

Dutzel, 2019-04-14:
Hello everyone, today I trained a net to recognize scissors. I set 30 epochs and only used the images from the fat dataset in the folder "037_scissors_16k". I hoped that at least something would be recognized. Unfortunately it was the same as @Abduoit with his first attempt, so that unfortunately nothing is recognized. On thursday I wanted to use the computers of my university, and see if I can recognize something with more computing power and more data. But I still have a few questions to make sure I'm doing it right:

1. is it correct, for example, to use the file net_epoch_30.pth as the weight that is generated by dope?
2. is the weight correctly specified in the config so follows?
```
weights: {"037_scissors_16k": "net_epoch_30.pth"}
dimensions:
{"037_scissors_16k": [20.259799957275391, 8.7530002593994141, 1.565500020980835]}
+ colors...
```
Or do I use wrong object names?
3. are there any further recommendations for the parameters to use?

If everything has gone through, I will report here gladly about the process. 

Thanks for your help :)

yuzw213, 2019-04-15:
hi，I tried 15,000 data on my computer, and I did 60 iterations, and I saw good results. Dope has a high requirement for diversity of data sets and needs to switch scenes and constantly change positions to collect data. There's nothing wrong with your settings, and 30 iterations should be enough for larger data objects

TontonTremblay, 2019-04-15:
@Dutzel: The version of FAT that I have has PNG, but when we saw how big the dataset was with PNGs we decided to move to JPEG, but internally I never made the switch, sorry about this. 

@yuzw213 and @Dutzel using about 200k images from FAT and DR I see 'ok' performance in the real world after 10 iterations, this is using a batchsize of 96 on 4 gpus.

Dutzel, 2019-04-17:
Thank you guys, I'll train as soon as I can get a free slot on our university renderfarm and report about my results.

Luedeke, 2019-04-28:
Hello,
the Average Distance Matrik (ADD) is not implemented in the train.py file or am I missing something?
And nothing is written in the test_metric.csv?

TontonTremblay, 2019-04-28:
Hello @Luedeke we do not provide evalutation code. We baised our implementation off his: https://github.com/thodan/obj_pose_eval

Luedeke, 2019-05-02:
Hi @TontonTremblay 
thank you so much for your help! This https://github.com/thodan/obj_pose_eval/blob/master/samples/eval_rotating_cup.py file was programmed for a 3D mesh model. You probably can't just apply this to the mesh?

Dutzel, 2019-05-02:
Hello, guys. I'm still trying to train a model with our university render farm. Unfortunately I am facing a new problem: 

 `start: 14:02:30.394044
load data
training data: 15375 batches
load models
ERROR: Unexpected bus error encountered in worker. This might be caused by insufficient shared memory (shm).
Traceback (most recent call last):
  File "train.py", line 1391, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1329, in _runnetwork
    for batch_idx, targets in enumerate(loader):
  File "/usr/local/lib/python2.7/dist-packages/torch/utils/data/dataloader.py", line 280, in __next__
    idx, batch = self._get_batch()
  File "/usr/local/lib/python2.7/dist-packages/torch/utils/data/dataloader.py", line 259, in _get_batch
    return self.data_queue.get()
  File "/usr/lib/python2.7/Queue.py", line 168, in get
    self.not_empty.wait()
  File "/usr/lib/python2.7/threading.py", line 340, in wait
    waiter.acquire()
  File "/usr/local/lib/python2.7/dist-packages/torch/utils/data/dataloader.py", line 178, in handler
    _error_if_any_worker_fails()
RuntimeError: DataLoader worker (pid 1157) is killed by signal: Bus error.
`
Do you have any idea what I can do? I've tried a lot, like adding --shm-size 50g to the nvidia-docker run command. But nothing helps.

TontonTremblay, 2019-05-02:
@Luedeke For debugging I have written a version that uses the 3d cuboid points. 
```
def ADDErrorCuboid(pose_gu, pose_gt, cuboid):
    """
        Compute the ADD error for a given cuboid. 
        pose_gu is the predicted pose as a matrix
        pose_gt is the ground thruth pose as a matrix
        cuboid is a Cuboid3D object (see inference/cuboid.py)
    """
    from scipy import spatial

    #obj = self.__obj_model
    vertices = np.array(cuboid._vertices)
    # print (vertices.shape)
    vertices = np.insert(vertices,3,1,axis=1)
    vertices = np.rot90(vertices,3)
    # print(vertices)

    obj = vertices
    pred_obj = np.matmul(pose_gu, obj)
    
    # obj = self.__obj_model
    # pred_obj = np.matmul(pose_gu, obj)
    # print (pred_obj)

    actual_obj = np.matmul(pose_gt, obj)
    #actual_obj = np.matmul(self.LtNT, actual_obj_l)
    #print("PREDICTED OBJECT\n", pred_obj)
    #print("ACTUAL OBJECT\n", actual_obj)   
    dist = spatial.distance.cdist(pred_obj.T, actual_obj.T, 'euclidean')
    true_dist = [dist[i][i] for i in range(len(dist))]
    #for i in range(len(true_dist)):
    #    if true_dist[i] >6000:
    #        print(i, true_dist[i])
    # print (true_dist)
    # raise()
    return np.mean(true_dist)
```

saratrajput, 2019-05-03:
@Dutzel A couple of things that might help: 
* Add --workers=1 argument to train.py 

* Find out which GPUs are not utilized using *nvidia-smi* and use them when using docker. For eg: 
```
NV_GPU=0,1 nvidia-docker run -it --shm-size 8G --network=host -v ${HOST_DIR}:${CONTAINER_DIR}:rw -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=unix${DISPLAY} --name=${CONTAINER_NAME} nvidia-dope:kinetic-v1 bash
```

[30 more replies]

closed 2019-05-21 by TontonTremblay