NVlabs/Deep_Object_Pose #15: Editing the code to run on a CPU ?

Mo-637, 2019-01-02:
Trying to run the code on a CPU machine for inference on a static image, I've disabled CuDNN and I've edited the code to transfer all the variables and tensors to the CPU for calculations. Only issue is the torchvision VGG19 pretrained model used in the original code, which seems to be trained using a GPU, how can I transfer this model amd it's weights to the CPU for inference ?

joaocabogon, 2019-03-19:
Hey, can you say exactly where you added that line? Was that enough to make it run correctly without any CUDA problems?

Mo-637, 2019-03-19:
If I remember correctly I added it after the imports in the dope.py script.
It was enough to make it run without any errors from cudnn or pytorch.
But inference speeds were extremely slow.
Maybe even averaging around a couple of minutes per frame.