NVlabs/Deep_Object_Pose #16: What is the really size of the processed image before putting into the network?

BIGJUN777, 2019-01-05:
Hello, gays!

These days, I was reading the `train.py` in order to find out the architecture of the neural network. From the code I find that the original image(640X480) will be processed before feeding into the net. However, It is a little bit hard for me to find out the size of the processed image. Can anyone tell me about that? I want to introduce this neural network to our group. 

Thank you!

TontonTremblay, 2019-01-06:
When training the image is set at 400x400 I think. The inference code does not change the size of the image.

TontonTremblay, 2019-01-06:
Check in the dope.py for transform, I think the resize is commented out. Otherwise please consult the pytorch transform documentation for Resize().

chibai, 2019-03-19:
> When training the image is set at 400x400 I think. The inference code does not change the size of the image.

I was wondering why in the code of training phase, the img size is 400x400, but in the 2.1 Network architecture of paper, "In our experiments, w = 640, h = 480 "

TontonTremblay, 2019-03-19:
The network is trained with 400x400 images but works on any image sizes afterwards since the network does have any fully connected layers.