NVlabs/Deep_Object_Pose #17: Train won't converge

filipetrocadoferreira, 2019-01-11:
I've trained for more than 60 epochs (200) and loss still didn't converge. The script in this repo would reproduce the shown results??

TontonTremblay, 2019-01-12:
You are correct about just taking into account the belief maps, I will update the code on Monday with the right loss. Sorry about this. 

For training data, in the paper I used about 220k dataset, 110k from DR data containing only images of the object of interest and 110k from FAT.

TontonTremblay, 2019-01-14:
I just updated the train.py file to include loss on affinities. Sorry about this, this is what I added: 
```
                for l in output_affinities: #output, each belief map layers. 
                    loss_tmp = ((l - target_affinity) * (l-target_affinity)).mean()
                    loss += loss_tmp 
````

filipetrocadoferreira, 2019-01-16:
It's converging now

Abduoit, 2019-01-23:
@filipetrocadoferreira could you please tell me if you have changed the learning rate ? how did it converge?

filipetrocadoferreira, 2019-01-23:
For a `batch_size=16` i use `--lr 0.0001`

But I would recommend to try some epochs in different rates. I got totally different results using adjacent rates.

Abduoit, 2019-03-21:
@filipetrocadoferreira 
what do you mean by trying some epochs in different rate, do you mean to try different values of learning rate ??