NVlabs/Deep_Object_Pose #18: Loss begins with 0.03 for untrained object

LutfuKorkmaz, 2019-01-16:
I would like to train a new object from the dataset FAT. I have started the train.py file with the updated loss and used the dataset 035_power_drill_16k for training. However, the loss begins in Epoch 1 witth 0.015 and reaches 0.0003 immediately. Then it stays around 0.0003.

I don't understand how an untrained network reaches loss 0.015 for a new object directly. What was the maximum/minimum loss for you before/after training?

filipetrocadoferreira, 2019-01-23:
Are you using the latest commit?

Abduoit, 2019-03-21:
@LutfuKorkmaz @filipetrocadoferreira 
I also got same loss with using the latest commit, the loss is not converging 
I am wondering if there's solution for this ?

TontonTremblay, 2019-05-22:
The learning rate might be set too high. I am not certain.

closed 2019-05-22 by TontonTremblay

mintar, 2019-06-19:
In case anybody is still interested, I've trained a model for the power drill object on the FAT dataset plus 60k domain randomization images that I created myself, and it's working beautifully.

You can download the model here: [035_power_drill_60.pth](https://mega.nz/#!grpCDY6Y!cxhxt76g78NY8kr17R5yosEUUUiuW9IMVDnRaOINShY)

Here are the parameters I used for training:

```bash
python /root/catkin_ws/src/dope/src/training/train.py \
    --data /root/dataset_ndss_035_power_drill/ \
    --object 035_power_drill_16k \
    --outf /root/dope_models/035_power_drill_16k_20190603 \
    --loginterval 1 --gpuids 0 1 2 --batchsize 72 --epochs 60
```

The only thing I had to experiment with was the batchsize, because it depends on the amount of GPU memory available. I simply reduced the batchsize until I got no more out-of-memory errors (3x NVidia Titan XP 12 GB cards).

SHZ2006, 2019-10-02:
> In case anybody is still interested, I've trained a model for the power drill object on the FAT dataset plus 60k domain randomization images that I created myself, and it's working beautifully.
> 
> You can download the model here: [035_power_drill_60.pth](https://mega.nz/#!grpCDY6Y!cxhxt76g78NY8kr17R5yosEUUUiuW9IMVDnRaOINShY)
> 
> Here are the parameters I used for training:
> 
> ```shell
> python /root/catkin_ws/src/dope/src/training/train.py \
>     --data /root/dataset_ndss_035_power_drill/ \
>     --object 035_power_drill_16k \
>     --outf /root/dope_models/035_power_drill_16k_20190603 \
>     --loginterval 1 --gpuids 0 1 2 --batchsize 72 --epochs 60
> ```
> 
> The only thing I had to experiment with was the batchsize, because it depends on the amount of GPU memory available. I simply reduced the batchsize until I got no more out-of-memory errors (3x NVidia Titan XP 12 GB cards).

Hi, I have trained for other objects in FAT dataset and my own also, but the result is nothing. Do you think the big dataset is the solution or not?
What about your learning rate?
Have you put other objects in single folder except power drill?
Have you put  the mixed folder too or deleted it?

Thanks in advance.

mintar, 2019-10-02:
My dataset structure looks like this:

```
/root/dataset_ndss_035_power_drill/
├── dr60k
│   ├── 000000.json
│   ├── 000000.png
│   ├── 000001.json
│   ├── 000001.png
│   ├──    [...]
│   ├── 059998.json
│   ├── 059998.png
│   ├── 059999.json
│   ├── 059999.png
│   ├── _camera_settings.json
│   └── _object_settings.json
├── mixed_kitchen_0
│   ├── 000000.left.depth.png
│   ├── 000000.left.jpg
│   ├── 000000.left.json
│   ├── 000000.left.seg.png
│   ├── 000000.right.depth.png
│   ├── 000000.right.jpg
│   ├── 000000.right.json
│   ├── 000000.right.seg.png
│   ├── 000001.left.depth.png
│   ├── 000001.left.jpg
│   ├── 000001.left.json
│   ├── 000001.left.seg.png
│   ├── 000001.right.depth.png
│   ├── 000001.right.jpg
│   ├── 000001.right.json
│   ├── 000001.right.seg.png
│   ├──    [...]
│   ├── 001999.left.depth.png
│   ├── 001999.left.jpg
│   ├── 001999.left.json
│   ├── 001999.left.seg.png
│   ├── 001999.right.depth.png
│   ├── 001999.right.jpg
│   ├── 001999.right.json
│   ├── 001999.right.seg.png
│   ├── _camera_settings.json
│   └── _object_settings.json
├── mixed_kitchen_1
├── mixed_kitchen_2
├── mixed_kitchen_3
├── mixed_kitchen_4
├── mixed_kitedemo_0
├── mixed_kitedemo_1
├── mixed_kitedemo_2
├── mixed_kitedemo_3
├── mixed_kitedemo_4
├── mixed_temple_0
├── mixed_temple_1
├── mixed_temple_2
├── mixed_temple_3
├── mixed_temple_4
├── single_kitchen_0
├── single_kitchen_1
├── single_kitchen_2
├── single_kitchen_3
├── single_kitchen_4
├── single_kitedemo_0
├── single_kitedemo_1
├── single_kitedemo_2
├── single_kitedemo_3
├── single_kitedemo_4
├── single_temple_0
├── single_temple_1
├── single_temple_2
├── single_temple_3
└── single_temple_4
    ├── 000000.left.depth.png
    ├── 000000.left.jpg
    ├── 000000.left.json
    ├── 000000.left.seg.png
    ├── 000000.right.depth.png
    ├── 000000.right.jpg
    ├── 000000.right.json
    ├── 000000.right.seg.png
    ├── 000001.left.depth.png
    ├── 000001.left.jpg
    ├── 000001.left.json
    ├── 000001.left.seg.png
    ├── 000001.right.depth.png
    ├── 000001.right.jpg
    ├── 000001.right.json
    ├── 000001.right.seg.png
    ├──     [...]
    ├── 000099.left.depth.png
    ├── 000099.left.jpg
    ├── 000099.left.json
    ├── 000099.left.seg.png
    ├── 000099.right.depth.png
    ├── 000099.right.jpg
    ├── 000099.right.json
    ├── 000099.right.seg.png
    ├── _camera_settings.json
    └── _object_settings.json
```

* The `dr60k` folder consists of the 60k domain randomized pictures I generated myself
* The `mixed_*` folders are directly taken from the FAT dataset
* The `single_*` folders are also from the FAT dataset, but only those for the object `035_power_drill`. In other words, I copied the folder `fat/single/035_power_drill_16k/kitchen_0/` from the FAT dataset into the folder `single_kitchen_0` in my dataset. This has the effect that my dataset folder only has a single level of subdirectories, but I believe this is not strictly necessary - I mainly wanted to exclude all the `single_*` folders for the other objects that I'm not training for.

> Hi, I have trained for other objects in FAT dataset and my own also, but the result is nothing. Do you think the big dataset is the solution or not?

You should definitely get some results without the domain randomization dataset. In the paper, they show that you can train on FAT only, DR only or FAT+DR. While FAT+DR generally gives you the best results, the other ones are often quite okay too, and never "nothing".

>  What about your learning rate?

For a different object, I've played around with the learning rate a lot (starting with a high rate, stopping the training, resuming with a lower rate and so on), but it turned out that it's not necessary at all. DOPE uses the Adam adaptive learning rate optimization algorithm for training, so it will automatically adjust the learning rate over time. As long as the loss goes down immediately after starting the training, I believe it's best to just leave the learning rate at the default and let Adam do its thing.

SHZ2006, 2019-10-04:
> My dataset structure looks like this:
>
> ```
> /root/dataset_ndss_035_power_drill/
> ├── dr60k
> │   ├── 000000.json
> │   ├── 000000.png
> │   ├── 000001.json
> │   ├── 000001.png
> │   ├──    [...]
> │   ├── [...]

Thanks a lot for your complete answer. As I understand in domain randomized pictures you just have image and .json and there isn't .depth and .seg, am I right?
Is it possible for you to upload somewhere a small part of your DR pictures?
Another question is about loss parameter. How much it should be low? For example within 10^-8 is ok? or maybe 10^-3?
Thanks in advance

mintar, 2019-10-07:
> Thanks a lot for your complete answer. As I understand in domain randomized pictures you just have image and .json and there isn't .depth and .seg, am I right?

Correct. DOPE only uses the RGB image, so the depth and segmentation images are ignored.

> Is it possible for you to upload somewhere a small part of your DR pictures?

I've uploaded the full dataset for you here:

[dr60k_00.tgz](https://mega.nz/#!17ZWUSqC!jGnaUg8aB9UbagoklFAlm-1pwBnSzHBqGdocW8f7cqs)
[dr60k_01.tgz](https://mega.nz/#!N2QWSIzb!zzKleiqriU0IKTvEcNWNYYxAxzwewyhbvdVxvrs-lDI)
[dr60k_02.tgz](https://mega.nz/#!hmYkSKjK!5rH7qG9yILySfpgQC2SYuZKBBPmoUshobmrS_qGA0WI)
[dr60k_03.tgz](https://mega.nz/#!IrYQmSjC!ALTPVayw5D8kK8jsxK6BqKzBFhfoZqsadov7S6U7ruY)
[dr60k_04.tgz](https://mega.nz/#!YrZWhIaI!625YZ3tJiNSJ5nhdMrPGtWJKBnxxlfnDR9S7uBMTWCE)
[dr60k_05.tgz](https://mega.nz/#!syQgQaDI!rtd-AKXCpZWWmze_nQ3qyYXEuGfHWlYd0UBXRnwEip8)

> Another question is about loss parameter. How much it should be low? For example within 10^-8 is ok? or maybe 10^-3?

The loss started at 0.03 and quickly went down. In the end, it was about 0.007. I still let it run for the full 60 epochs because a low training loss doesn't automatically mean that the training has finished - you'd have to do testing in between, and I didn't do that because of some issues.

SHZ2006, 2019-10-10:
> Correct. DOPE only uses the RGB image, so the depth and segmentation images are ignored.
>
>
> I've uploaded the full dataset for you here: [...]

Thank you so much. I will try and say the result here as well.