NVlabs/Deep_Object_Pose #20: Rotation Axes

aditya2592, 2019-02-05:
Is it possible to know the exact orientation of the axes in with respect to which we get the rotation values during inference ?

TontonTremblay, 2019-05-22:
I am not sure I understand the question, could give me an example?

aditya2592, 2019-05-22:
Hi @TontonTremblay 
I wanted to ask if the rotations in the annotated FAT dataset and those predicted by DOPE are in the camera frame or world frame

TontonTremblay, 2019-05-22:
I am not sure, this should be quite easy to double check, just run DOPE with http://wiki.ros.org/image_publisher feeding FAT images and check the quaternion. I would not be surprised if you need to apply a 90 degree rotation.

closed 2019-05-22 by TontonTremblay

mintar, 2019-05-24:
The poses are in the camera frame.

aditya2592, 2019-05-24:
Thanks @mintar. We are also given a camera pose in the world frame in the FAT dataset. So if I need the object pose in world frame, I can get it just by multiplying object pose in camera frame by the camera pose in world frame right?

mintar, 2019-05-30:
Right! :)

aditya2592, 2019-05-31:
Is there an expected orientation of the worldframe in FAT images with respect to gravity? Like will z be pointing upwards from the ground or be parallel to the ground? Or can the worldframe can be any random point with a random orientation in the scene?
I am trying to visualize poses in the worldframe of FAT images and wanted to make sure I am getting the right results.

UniqueVN, 2019-05-31:
The FAT dataset use OpenCV coordinate system: Z - forward, X - right, Y - down. You can see the detail here: http://homepages.inf.ed.ac.uk/rbf/CVonline/LOCAL_COPIES/OWENS/LECT9/node2.html

aditya2592, 2019-05-31:
So that means in all scenes, Y axis would be perpendicular to the ground and pointing downwards?

UniqueVN, 2019-05-31:
Yes, that's right. In the object's location, if Y is > 0, it mean the object is below the center of the view.

IbrahimMCode, 2020-03-31:
Hello, 
Is it possible to visualize the Rotation Axes with the predicted DOPE Bounding boxes similar to the Figure3 in the Paper?
Thank you