NVlabs/Deep_Object_Pose #21: In training, there are some hardware problems.

xdw-123, 2019-02-12:
I can identify the object I want in data set and training, but I want to transplant it to TX2 and find that the camera recognition process is too late. Have you met it? I suspect that the GPU acceleration is not turned on because I can't find the status of the GPU in the terminal.

TontonTremblay, 2019-02-12:
We have run internally DOPE on TX2 and it does work but lower framerate than the one reported on Titan X.

xdw-123, 2019-02-13:
I am the same, but my computer configuration is graphics card: GTX1080, processor: i7-7700. My computer identification time delay is almost small, TX2 work delay to about 1 minute. Is there such a big delay?Is there any way to improve it?

xdw-123, 2019-02-13:
Thank you for calling back.I am the same, but my computer configuration is graphics card: GTX1080, processor: i7-7700. My computer identification time delay is almost small, TX2 work delay to about 1 minute. Is there such a big delay?Is there any way to improve it?

xdw-123, 2019-02-14:
Thank you for calling back.I am the same, but my computer configuration is graphics card: GTX1080, processor: i7-7700. My computer identification time delay is almost small, TX2 work delay to about 5s. Is there such a big delay?Is there any way to improve it?

TontonTremblay, 2019-05-22:
You could improve it by using tensor RT. You could also write your own DOPE cuda version, but that would take some time. You could use the C++ pytorch API. You can also use the output from stage 1. In the file `detector.py` change the `stop_at_stage` value to 1 or 2. This is set on line 59. This will improve the performance, but the poses wont be as good.

closed 2019-05-22 by TontonTremblay