NVlabs/Deep_Object_Pose #22: Ask you about the details of the data set.

xdw-123, 2019-02-18:
I can now detect what I want, but the high reflective metal object in the frame structure is white in UE4, which does not match the color of the real object. I think you mentioned this problem in your paper. Is there a better solution at present?

TontonTremblay, 2019-02-19:
I suggest looking into this: https://www.youtube.com/watch?v=sTY0TnDJjYs 

You should also randomized the material properties using RandomParams in NDDS.

closed 2019-05-22 by TontonTremblay

dj4park, 2019-07-29:
@xdw-123 Hello, I am trying to implement DOPE to determine the pose of a glass bottle that is not part of the YCB dataset. Could you outline the process of how you trained your custom dataset? I only have a .obj file available for my bottle that I downloaded from [here](https://sketchfab.com/3d-models/soju-bottle-efefa415042b4a23a2cf397c216b6754). 

Your help is greatly appreaciated! Thank you.