NVlabs/Deep_Object_Pose #23: dope crashing on oom during inference not training

rohitmenon86, 2019-02-21:
I have an Ubuntu 16.04 laptop with GeForce 940MX GPU. I am currently publishing a single image of the sugar box from the YCB dataset as I don't have the real object at hand. I am loading only the sugar_60.pth weight currently.
When I run the dope node, it crashes with an Out of Memory issue
THCudaCheck FAIL file=/pytorch/aten/src/THC/generic/THCStorage.cu line=58 error=2 : out of memory

As I understand, this could be related to the batch size. However will it affect the inferencing as well? How to resolve this error

rohitmenon86, 2019-02-21:
PS I am a pytorch noob. I am into manipulation and grasping and this is my first foray into using pytorch based learning library. Maybe this has a quick fix but I am not aware. So, apologies in advance

TontonTremblay, 2019-02-25:
I never tried running dope on something else than a 1060 (laptop). Do you see the gpu when you type `nvidia-smi` in the terminal. Are you sure you are not using the internal display. Anyway just to say that gpus in laptops are very clumsy in general.

closed 2019-05-22 by TontonTremblay