NVlabs/Deep_Object_Pose #25: Error with CUDA when running the dope.py

joaocabogon, 2019-03-18:
Hey, when I run the dope.py script I get the following error, I don't have a GPU and I was trying to make it work with the image of the cracker box in Gazebo below: 

LoadingDOPE parameters from '/home/joao/atrv_ws/src/dope/config/config_pose.yaml'...
    Parameters loaded.
Loading DOPE model '/home/joao/atrv_ws/src/dope/weights/cracker_60.pth'...
THCudaCheck FAIL file=/pytorch/aten/src/THC/THCGeneral.cpp line=70 error=30 : unknown error
Traceback (most recent call last):
  File "/home/joao/atrv_ws/src/dope/src/dope.py", line 272, in <module>
    run_dope_node(params)
  File "/home/joao/atrv_ws/src/dope/src/dope.py", line 150, in run_dope_node
    models[model].load_net_model()
  File "/home/joao/atrv_ws/src/dope/src/inference/detector.py", line 228, in load_net_model
    self.net = self.load_net_model_path(self.net_path)
  File "/home/joao/atrv_ws/src/dope/src/inference/detector.py", line 239, in load_net_model_path
    net = torch.nn.DataParallel(net, [0]).cuda()
  File "/home/joao/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 249, in cuda
    return self._apply(lambda t: t.cuda(device))
  File "/home/joao/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 176, in _apply
    module._apply(fn)
  File "/home/joao/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 176, in _apply
    module._apply(fn)
  File "/home/joao/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 176, in _apply
    module._apply(fn)
  File "/home/joao/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 182, in _apply
    param.data = fn(param.data)
  File "/home/joao/.local/lib/python2.7/site-packages/torch/nn/modules/module.py", line 249, in <lambda>
    return self._apply(lambda t: t.cuda(device))
RuntimeError: cuda runtime error (30) : unknown error at /pytorch/aten/src/THC/THCGeneral.cpp:70

[image]

TontonTremblay, 2019-03-18:
You can for sure run it on CPU although the convention for using data parallel in Pytorch is a little weird as it adds a prefix `module` to the save dict weights. Please refer to this to load the weights: https://github.com/pytorch/pytorch/issues/9176. You can also do something like this: 
```python
# original saved file with DataParallel
state_dict = torch.load('myfile.pth.tar')
# create new OrderedDict that does not contain `module.`
from collections import OrderedDict
new_state_dict = OrderedDict()
for k, v in state_dict.items():
    name = k[7:] # remove `module.`
    new_state_dict[name] = v
# load params
model.load_state_dict(new_state_dict)
```
from https://discuss.pytorch.org/t/solved-keyerror-unexpected-key-module-encoder-embedding-weight-in-state-dict/1686/3

joaocabogon, 2019-03-19:
can you please be more specific on where to include those lines?

TontonTremblay, 2019-05-21:
Closing since no activities.

closed 2019-05-21 by TontonTremblay