NVlabs/Deep_Object_Pose #26: problem on creating ground truth of new dataset.

chibai, 2019-03-19:
Dear TontonTremblay:
I read the paper very carefully and hope to organize my own dataset for training, but I do have problem about setting the ground truth of my own dataset.

    According the paper, there're 2 outputs of the network, 9 belief maps and 16 vector fields.  For the belief maps, I fully understand how to set ground truth, which represents 8 vertices and 1 centroid of 3D bounding box, where do a Gaussian filter with   = 2 pixels.

But for the vector fileds, I'm very confused. "The ground truth vector fields were generated by setting pixels to the normalized x- and y-components of the vector pointing toward the object’s centroid. Only the pixels within a 3-pixel radius of each groundtruth vertex were set in this manner, with all other pixels set to zero. Wherever more than one vertex resided within this radius, one of them was selected at random to be used for generating these components." 

As far as I knew, 16 vector fileds relate to 8 vertices of 3D bouding box.  Let's say vectorFiled1(x component) and vectorFiled2(y component) relate to vertex1.

Let's assume in the ground truth map with size (50, 50), vertex1 is (0, 0) and centroid is (10, 10), what is the vectorFiled1 and vectorFiled2 ground truth like???

BEST WISHES!
                                                                                     a very worry person

TontonTremblay, 2019-03-19:
If you look at the `train.py` file there is a function that I copied here:

```python
def GenerateMapAffinity(img,nb_vertex,pointsInterest,objects_centroid,scale):
    """
    Function to create the affinity maps, 
    e.g., vector maps pointing toward the object center. 
    Args:
        img: PIL image
        nb_vertex: (int) number of points 
        pointsInterest: list of points 
        objects_centroid: (x,y) centroids for the obects
        scale: (float) by how much you need to scale down the image 
    return: 
        return a list of tensors for each point except centroid point      
    """

    # Apply the downscale right now, so the vectors are correct. 
    img_affinity = Image.new(img.mode, (int(img.size[0]/scale),int(img.size[1]/scale)), "black")
    # Create the empty tensors
    totensor = transforms.Compose([transforms.ToTensor()])

    affinities = []
    for i_points in range(nb_vertex):
        affinities.append(torch.zeros(2,int(img.size[1]/scale),int(img.size[0]/scale)))
    
    for i_pointsImage in range(len(pointsInterest)):    
        pointsImage = pointsInterest[i_pointsImage]
        center = objects_centroid[i_pointsImage]
        for i_points in range(nb_vertex):
            point = pointsImage[i_points]
            affinity_pair, img_affinity = getAfinityCenter(int(img.size[0]/scale),
                int(img.size[1]/scale),
                tuple((np.array(pointsImage[i_points])/scale).tolist()),
                tuple((np.array(center)/scale).tolist()), 
                img_affinity = img_affinity, radius=1)

            affinities[i_points] = (affinities[i_points] + affinity_pair)/2

            # Normalizing
            v = affinities[i_points].numpy()                    
            
            xvec = v[0]
            yvec = v[1]

            norms = np.sqrt(xvec * xvec + yvec * yvec)
            nonzero = norms > 0

            xvec[nonzero]/=norms[nonzero]
            yvec[nonzero]/=norms[nonzero]

            affinities[i_points] = torch.from_numpy(np.concatenate([[xvec],[yvec]]))
    affinities = torch.cat(affinities,0)

return affinities
```
That functions generates the affinity maps. I hope this help. The maps are generated in the original image size then they are cropped and scaled.

chibai, 2019-03-20:
Thanks a lot, the code can solve my problem

closed 2019-03-20 by chibai

chibai, 2019-03-20:
> If you look at the `train.py` file there is a function that I copied here:
>
> ```python
> def GenerateMapAffinity(img,nb_vertex,pointsInterest,objects_centroid,scale):
>     """
>     Function to create the [...]

Thanks for your answering, I read the code carefully about how to create ground truth, but do have a problem about the training code, since it's not supported, I'm fine without answers

I didn't see any use of the code affinities[i_points] = (affinities[i_points] + affinity_pair)/2 in your quote. Cause affinities[i_points] must be zero matrix by affinities.append(torch.zeros(2,int(img.size[1]/scale),int(img.size[0]/scale)))
and affinity_pair already saved the normalized x,y component by getAfinityCenter()