NVlabs/Deep_Object_Pose #27: Backup download url for FAT dataset is begged

chibai, 2019-03-20:
Because the Google is blocked by China's government, I can't really download the FAT dataset successfully.  Is there anyone can offer a backup download url for Chinese users??May be drop-box or other drives?

I am begging anyone who passed by could send me one object file to test the training process.
My email is 470598003@qq.com

Abduoit, 2019-03-21:
I ll send you one object now

chibai, 2019-03-22:
> I ll send you one object now

Thanks for you power drill, but you still put your dataset on the Google drive. 
It seems your dataset url is private and I can not download directly.
And I already apply the permission of your power drill dataset

Abduoit, 2019-03-25:
@chibai  are you able to get the data now ??

chibai, 2019-04-08:
> @chibai are you able to get the data now ?? I gave you an access

Thanks， I got full dataset through my classmate, who is working in America

closed 2019-05-21 by TontonTremblay

571502680, 2019-06-16:
> @chibai are you able to get the data now ?? I gave you an access
> 
> Thanks， I got full dataset through my classmate, who is working in America

Could you share with this datasets？

mintar, 2019-06-18:
@TontonTremblay : Perhaps it would be a good idea to upload a copy of the FAT dataset here: https://zenodo.org/

It's a free and open hosting platform for research outputs, and I've been quite happy with it. Some features:

* *free* (funded by a European Union Open Access initiative)
* runs on CERN infrastructure - compared to what they count as "big data", our datasets are tiny :)
* up to 50 GB
* CERN commits to storing the data for at least 20 years, so I trust Zenodo more than some Google Drive link that might be gone any day when the author needs to free up disk space
* you even get a citable DOI for it

Since the data is immutable once uploaded, it's a good idea to play around with it first on their sandbox instance: https://sandbox.zenodo.org/