NVlabs/Deep_Object_Pose #29: Custom Objects

MohamadJaber1, 2019-04-04:
Firstly, thank you so much fro publishing and sharing this amazing work!
I am doing my master thesis in Germany, and I want to find different scattered objects in the robotic simulation environment (RobotStudio) and then grasp and put them in their right positions.

I already have the 3D CD models. I implemented some papers to get the 6D pose. But what I knew so far from your work is that DOPE find the objects coordinates with respect to the camera. And also you are using a robotic manipulation in your work which is even more interesting.

Now, I haven't implemented DOPE yet since I want to know about the steps I shall follow to follow to be able to find the 6D pose of my own objects in my simulated environment. Also, can I use Windows? or Ubuntu is a required dependency?. I would really appreciate your reply. DOPE is a great work to implement. And Thank you.

TontonTremblay, 2019-04-04:
Hello, 

In order to train for your own objects you need to invest some time in generating the data using the NDDS (https://github.com/NVIDIA/Dataset_Synthesizer) then once you have the data, training DOPE is straight forward. 

You can run the inference on windows. You can also generate the data on windows, the original data was generated on a windows machine. But training will have to occur on ubuntu with at least one GPU.

MohamadJaber1, 2019-04-05:
Thank you @TontonTremblay for this instant reply. I do have one strong GPU. For Ubuntu, I think I will use VirtualBox. 
I am looking forward for implementing DOPE and showing the results

closed 2019-05-21 by TontonTremblay

TDBTECHNO, 2019-06-10:
But how to label for the custom object dataset?

TrinhTUHH, 2019-10-03:
@TontonTremblay  If I have only one custom object, how much data do you think I should generate?

TontonTremblay, 2019-10-08:
I would start with 20k. But be careful with symmetries on your object.

TrinhTUHH, 2019-10-08:
@TontonTremblay in the paper you mentioned 120K (60K domain randomization and 60K photorealistic) for each object and mixed object cases, got the best result. Is 20K kind of too less?
If my object is symmetric, then what should I take care of when generate data?

TontonTremblay, 2019-10-09:
You need to be careful which poses you show the training algorithm. Make sure you only show the same face in the same way, e.g., only showing the front face of the object even though the back is the same visually. The way I do it in NDDS is to fix the pose of the object and limit the camera's movement. 

I have noticed that 20k of DR images works fairly well as a first prototype. Then you can generate more.