NVlabs/Deep_Object_Pose #32: Train on Custom NDDS Objects

Luedeke, 2019-04-09:
Hello everybody,

I created a lot of data with NDDS, now I wanted to train it, but:

```
/catkin_ws/src/dope/src/training$ python train.py --data /media/username/Ubuntu-TMP/Stereo_Dataset/ --object Objctname --outf Objctname 
start: 17:22:19.190897
load data
training data: 11375 batches
load models
Traceback (most recent call last):
  File "train.py", line 1391, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1329, in _runnetwork
    for batch_idx, targets in enumerate(loader):
  File "/home/username/catkin_ws/local/lib/python2.7/site-packages/torch/utils/data/dataloader.py", line 286, in __next__
    return self._process_next_batch(batch)
  File "/home/username/catkin_ws/local/lib/python2.7/site-packages/torch/utils/data/dataloader.py", line 307, in _process_next_batch
    raise batch.exc_type(batch.exc_msg)
ValueError: Traceback (most recent call last):
  File "/home/username/catkin_ws/local/lib/python2.7/site-packages/torch/utils/data/dataloader.py", line 57, in _worker_loop
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "train.py", line 446, in __getitem__
    data = loader(txt, self.objectsofinterest,img)
  File "train.py", line 276, in loadjson
    data = json.load(data_file)
  File "/usr/lib/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

```

What did I do wrong?
Thank you

saratrajput, 2019-04-10:
Do you have a file named _object_settings.json in your folder?

yuzw213, 2019-04-10:
hi，I looked at your JSON file. You added two objects to the scene. Training should specify the class name of the object you want to train, such as "FerreroKuesschen"

Luedeke, 2019-04-10:
@saratrajput  Hi, yes I have one. I attached it.

@yuzw213  Hi, shit I forgot to delete that one. But you can train on mixed Objects? Tthan they are also in the number.left.json File? I added some Data of my Mixed Directory.
[Mixed.tar.gz](https://github.com/NVlabs/Deep_Object_Pose/files/3064212/Mixed.tar.gz)

Or you mean: /catkin_ws/src/dope/src/training$ python train.py --data /media/username/Ubuntu-TMP/Stereo_Dataset/ --object FerreroKuesschen --outf FerreroKuesschen 
I tried that, that was not working :/ same error.

yuzw213, 2019-04-11:
@Luedeke I haven't encountered this error. It is OK to add a "mixed" directory, but each frame contains the best file format like the "fat" data set. Each frame data should contain 8 files, 4 left and right, respectively. Rgb diagram, depth diagram, Split the graph and JSON files, you can try it out, see if it works

saratrajput, 2019-04-11:
I suspect the problem might be here. You say you are trying to train multiple classes at the same time but your arguments for training specify a single object which by the way is not present in your *_object_settings.json* file. I haven't tried training multiple classes at the same time yet, so maybe try training them separately and provide the weights for different objects during inference in the configuration file. 

```python
python train.py --data /media/username/Ubuntu-TMP/Stereo_Dataset/ --object Objctname --outf Objctname
```

Luedeke, 2019-04-11:
The problem was the Dataset, its now working with another Object.
Thank you!!!

Luedeke, 2019-04-11:
@yuzw213  Could you send me an e-mail? I would like to ask you a few more questions.

Dutzel, 2019-04-11:
How does your file structure look like? I am currently trying to get into training, but I allways get the ValueError: num_samples should be a positive integeral value, but got num_samples=0
... after e.g. python train.py --data /home/dustin/fat/ --object 003_cracker_box_16k --outf 003_cracker_box_16k --gpuids 0

any ideas?

could you send me your ferrero küsschen dataset so I can try it out with your data?

Luedeke, 2019-04-11:
@Dutzel My file Structure in this test was: Dataset/FerreroKuesschen
and i tested it with: 
`python train.py --data /media/username/Ubuntu-TMP/Stereo_Dataset/ --object FerreroKuesschen --outf FerreroKuesschen`

I try to upload it .

Do you know anything about transfer learning?

Dutzel, 2019-04-12:
@Luedeke I allready found the solution. My problem was, that the fat dataset uses jpgs instead of png, and I assumed the dataset is adjusted for this repo.

Have you had problems with the "RuntimeError: CUDA out of memory"?

Luedeke, 2019-04-12:
@Dutzel great.
Something similar yes, you have to reduce the batch size, it should be 32. Try it with 16.

Dutzel do you know something about Transfer Learning?

Dutzel, 2019-04-12:
@Luedeke  thanks.
I allready tryed several batchsizes and 12 works now, but I'm still unhappy with it...

Unfortunately I had nothing to do with transfer learning in a deep learning context. I myself am still really new in this area and am trying to get used to it.

Luedeke, 2019-04-12:
@Dutzel Well, I train on a GTX 1070, so 16 are possible maybe even more. But now we have to use a higher learning speed or something like that.  I'm not sure what exactly. I also need to understand that and find more literature on it.

I'm also new in this area so I can't help much. But if we have a similar goal, we could exchange ideas.

TontonTremblay, 2019-04-15:
@Luedeke I think you original error comes from a missing json file you generated with ndds. Can you check if the file you are trying to load exists? Moreover, you should be able to train on multiple objects, normally if you don't specify the object it should load all the classes present and treat them as one class. 

You could also change the optimizer for SGD with a scheduler, this would use less memory than adam and thus you could increase the batchsize.

Guptajakala, 2019-05-16:
@TontonTremblay Is DOPE able to train multi-instance same class objects in one image?

TontonTremblay, 2019-05-16:
@Guptajakala yes it was designed exactly for this :P

closed 2019-05-21 by TontonTremblay

Gaoee, 2021-01-14:
Hi, I am also trying to use NDDS to generate a new data set. But how can I import my FBX format 3D model and replace the original object in NDDS.
Thank you.