NVlabs/Deep_Object_Pose #33: error:    nvidia-docker build -t nvidia-dope:kinetic-v1 -f Dockerfile.kinetic ..

zkl99999, 2019-04-10:
Downloading https://files.pythonhosted.org/packages/df/a4/7f5ec6e9df1bf13f1881353702aa9713fcd997481b26018f35e0be85faf7/torch-0.4.0-cp27-cp27mu-manylinux1_x86_64.whl (484.0MB)  

File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/response.py", line 240, in _error_catcher
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.
You are using pip version 8.1.1, however version 19.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The command '/bin/sh -c pip install --no-cache-dir -r requirements.txt' returned a non-zero code: 2

saratrajput, 2019-04-11:
This might be a network problem.

TontonTremblay, 2019-04-15:
Can you provide more information about the error. When are you getting this error?

TontonTremblay, 2019-05-21:
I am closing this error since there are no activities.

closed 2019-05-21 by TontonTremblay