NVlabs/Deep_Object_Pose #35: Fixing-Errors-Summary

Dutzel, 2019-04-12:
So that not everyone who wants to take an example from this project has to read cross-reading to get the project going, here is a short summary:

## PyTorch Version Error for RTX 2080 (CUDA 10 is also working)
Found GPU0 GeForce RTX 2080 which requires CUDA_VERSION >= 9000 for
optimal performance and fast startup time, but your PyTorch was compiled
with CUDA_VERSION 8000. 
### Fix
1. go went to: https://pytorch.org/get-started/previous-versions/
2. download the version you need (in my case it was) "cu90/torch-1.0.0-cp27-cp27mu-linux_x86_64.whl" - for python 2.7
3. go to downloads and typed "pip install cu90/torch-1.0.0-cp27-cp27mu-linux_x86_64.whl"

## Training your data Num_Samples = 0 
ValueError: num_samples should be a positive integeral value, but got num_samples=0
### Fix
Make sure you are using pngs instead of jpgs. (The fat dataset contains jpgs!)
Open the train.py and change the code e.g.:
#### PNG
```
    def add_json_files(path,):
        for imgpath in glob.glob(path+"/*.png"):
            if exists(imgpath) and exists(imgpath.replace('png',"json")):
                imgs.append((imgpath,imgpath.replace(path,"").replace("/",""),
                    imgpath.replace('png',"json")))
```
#### JPG
```
    def add_json_files(path,):
        for imgpath in glob.glob(path+"/*.jpg"):
            if exists(imgpath) and exists(imgpath.replace('jpg',"json")):
               imgs.append((imgpath,imgpath.replace(path,"").replace("/",""),
                  imgpath.replace('jpg',"json")))
```

## Index Error 
IndexError: invalid index of a 0-dim tensor. Use tensor.item() to convert a 0-dim tensor to a Python number
### Fix
In the train.py change all "loss.data[0]" to "loss.item()"

Dutzel, 2019-04-12:
I am still facing the "RuntimeError: CUDA out of memory. Tried to allocate 19.62 MiB (GPU 0; 7.76 GiB total capacity; 5.78 GiB already allocated; 11.00 MiB free; 54.01 MiB cached)"

After solving the problem, I will add the solution here too.

TontonTremblay, 2019-05-21:
The code was written to work with pytorch 0.4.1, it will have some issues with 1.0 which you need with the latest 2080. FYI I trained DOPE on 4 or 8 GPUs machine, so you will need a smaller batchsize to train on a single GPU.

closed 2019-05-21 by TontonTremblay