NVlabs/Deep_Object_Pose #36: Getting error on Ubuntu 16

michaelchi08, 2019-04-20:
michael@michael-Precision-7520:~/catkin_ws/src/dope/docker$ ./run_dope_docker.sh 
Container name     : nvidia-dope-v1
Host directory     : /home/michael/catkin_ws/src/dope
Container directory: /root/catkin_ws/src/dope
Creating new DOPE docker container.
access control disabled, clients can connect from any host
**Unable to find image 'nvidia-dope:kinetic-v1' locally**
docker: Error response from daemon: pull access denied for nvidia-dope, repository does not exist or may require 'docker login'.
See 'docker run --help'.

saratrajput, 2019-04-22:
Did you build the image first?

michaelchi08, 2019-04-24:
@saratrajput  This is the error i get when i try to build. Thanks for your help.

michael@michael-Precision-7520:~/catkin_ws/src/dope/docker$ nvidia-docker build -t nvidia-dope:kinetic-v1 -f Dockerfile.kinetic ..
Sending build context to Docker daemon  1.207GB
Step 1/21 : FROM nvidia/cudagl:9.0-devel-ubuntu16.04
 ---> 394999b1b6a7
Step 2/21 : ENV HOME /root
 ---> Using cache
 ---> cc9485b6e74c
Step 3/21 : WORKDIR ${HOME}
 ---> Using cache
 ---> 9b0ad89da35a
Step 4/21 : RUN apt-get update && apt-get -y  --no-install-recommends install software-properties-common
 ---> Using cache
 ---> 837537437e4f
Step 5/21 : RUN apt-get update && apt-get -y --no-install-recommends install         ant         bzip2         ca-certificates         ccache         cmake         curl         genromfs         git         gosu         iproute         iputils-ping         less         lcov         libcudnn7=7.1.4.18-1+cuda9.0         libcudnn7-dev=7.1.4.18-1+cuda9.0         libeigen3-dev         libopencv-dev         make         nano         net-tools         ninja-build         openjdk-8-jdk         patch         pkg-config         protobuf-compiler         python-argparse         python-dev         python-empy         python-numpy         python-pip         python-serial         python-software-properties         rsync         s3cmd         software-properties-common         sudo         unzip         xsltproc         wget         zip     && apt-get -y autoremove     && apt-get clean autoclean     && pip install setuptools wheel     && pip install 'matplotlib==2.2.2' --force-reinstall     && pip install cpp-coveralls     && pip install jinja2     && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*
 ---> Using cache
 ---> 459fefb215b1
Step 6/21 : WORKDIR ${HOME}
 ---> Using cache
 ---> 2e4a4f3bce96
Step 7/21 : RUN apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116     && sh -c 'echo "deb http://packages.ros.org/ros/ubuntu xenial main" > /etc/apt/sources.list.d/ros-latest.list'     && sh -c 'echo "deb http://packages.ros.org/ros-shadow-fixed/ubuntu/ xenial main" > /etc/apt/sources.list.d/ros-shadow.list'     && apt-get update && apt-get -y --no-install-recommends install         ros-kinetic-gazebo-ros-pkgs         ros-kinetic-mavros                  ros-kinetic-mavros-extras           ros-kinetic-ros-base                ros-kinetic-rviz                    ros-kinetic-tf2                     ros-kinetic-cv-bridge           && apt-get -y autoremove            && apt-get clean autoclean          && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*
 ---> Using cache
 ---> 6b4e70afe93e
Step 8/21 : RUN geographiclib-get-geoids egm96-5     && rosdep init                       && rosdep update
 ---> Using cache
 ---> 920debf4062f
Step 9/21 : RUN echo 'source /opt/ros/kinetic/setup.bash' >> ${HOME}/.bashrc
 ---> Using cache
 ---> 6aae0a7e472c
Step 10/21 : WORKDIR ${HOME}
 ---> Using cache
 ---> ac8a60759a74
Step 11/21 : RUN git clone http://github.com/opencv/opencv.git && cd opencv     && git checkout 3.3.1                       && mkdir build && cd build                  && cmake -D CMAKE_BUILD_TYPE=RELEASE            -D CMAKE_INSTALL_PREFIX=/usr/local          -D WITH_CUDA=OFF                             -D WITH_OPENCL=OFF                          -D ENABLE_FAST_MATH=1                       -D CUDA_FAST_MATH=1                         -D WITH_CUBLAS=1                            -D BUILD_DOCS=OFF                           -D BUILD_PERF_TESTS=OFF                     -D BUILD_TESTS=OFF                          ..                                      && make -j `nproc`                          && make install                             && cd ${HOME} && rm -rf ./opencv/
 ---> Using cache
 ---> f18a2c62d221
Step 12/21 : ENV CATKIN_WS ${HOME}/catkin_ws
 ---> Using cache
 ---> 2aa4650206b0
Step 13/21 : COPY docker/init_workspace.sh ${HOME}
 ---> Using cache
 ---> d346cf0bb769
Step 14/21 : RUN ${HOME}/init_workspace.sh
 ---> Using cache
 ---> a744ac13e279
Step 15/21 : ENV CCACHE_CPP2=1
 ---> Using cache
 ---> efadc5b4e24b
Step 16/21 : ENV CCACHE_MAXSIZE=1G
 ---> Using cache
 ---> 5e30b03ff272
Step 17/21 : ENV DISPLAY :0
 ---> Using cache
 ---> 67221ea97584
Step 18/21 : ENV TERM=xterm
 ---> Using cache
 ---> 80925663ccc0
Step 19/21 : ENV QT_X11_NO_MITSHM 1
 ---> Using cache
 ---> c0db5b755a77
Step 20/21 : COPY requirements.txt ${HOME}
 ---> Using cache
 ---> 0ed1906d6495
Step 21/21 : RUN pip install --no-cache-dir -r requirements.txt
 ---> Running in 0a875eb1d114
Collecting pyrr==0.9.2 (from -r requirements.txt (line 1))
  Downloading https://files.pythonhosted.org/packages/24/67/707ea36bdd37415020bec41f571b7f08dfb017c2b741b973b589bb9ef71d/pyrr-0.9.2-py2.py3-none-any.whl (65kB)
Collecting torch==0.4.0 (from -r requirements.txt (line 2))
Exception:
Traceback (most recent call last):
[... 7 lines cut ...]
    session=self.session, hashes=hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 810, in unpack_url
    hashes=hashes
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 649, in unpack_http_url
    hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 842, in _download_http_url
    stream=True,
  File "/usr/share/python-wheels/requests-2.9.1-py2.py3-none-any.whl/requests/sessions.py", line 480, in get
    return self.request('GET', url, **kwargs)
[... cut ...]
  File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 610, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 228, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
You are using pip version 8.1.1, however version 19.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The command '/bin/sh -c pip install --no-cache-dir -r requirements.txt' returned a non-zero code: 2

saratrajput, 2019-04-24:
This might be simply a network problem. Try again. You didn't change anything in the Dockerfile right?

michaelchi08, 2019-04-24:
I didn't change it and i'm guessing it didn't build properly then.  The error seem to be coming from python within the docker image?

"TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'"

TontonTremblay, 2019-04-24:
This happens when installing torch. Can you try to install torch without the docker image on you machine. I checked online and there might be a problem with pip itself, but it does not make sense since it is in the docker container and I should have the same problem locally. 

Can you tell me more about your machine please? e.g. hardware and OS running?

saratrajput, 2019-04-25:
I still think it's a network connectivity issue. Are you behind a proxy? Or try again when your internet connection is better. Here are a couple of discussions with same error for different packages:

https://github.com/pypa/pip/issues/4779

https://stackoverflow.com/questions/42610545/typeerror-unsupported-operand-types-for-retry-and-int-during-pip

closed 2019-05-21 by TontonTremblay