NVlabs/Deep_Object_Pose #38: Nvdu YCB setup fails with 'NoneType' object has no attribute 'obj_settings'

nouyang, 2019-05-01:
Reporting here due to lack of issues tab on nvdu proper,
Running `nvdu_ycb --setup` gave me 

```
AttributeError: 'NoneType' object has no attribute 'obj_settings'` 
```

In order to get it to work, I followed instructions at 
https://zenodo.org/record/2579173
and used a fork of the nvdu repository
```
$ virtualenv -p python3.6 ~/venv3.6
$ cd venv_nvdu/
$ source bin/activate
$ pip install -e 'git+https://github.com/mintar/Dataset_Utilities.git#egg=nvdu'
$ nvdu_ycb -s
```

Full error:
```
(mlenv) rui@rui-chai:~/Documents/6dof_project/dope$ (master)$ nvdu_ycb --setup
Traceback (most recent call last):
  File "/home/rui/mlenv/bin/nvdu_ycb", line 10, in <module>
    sys.exit(main())
  File "/home/rui/mlenv/lib/python3.7/site-packages/nvdu/tools/nvdu_ycb.py", line 237, in main
    setup_all_ycb_models()
  File "/home/rui/mlenv/lib/python3.7/site-packages/nvdu/tools/nvdu_ycb.py", line 214, in setup_all_ycb_models
    for obj_name, obj_settings in all_ycb_object_settings.obj_settings.items():
AttributeError: 'NoneType' object has no attribute 'obj_settings'
```

thangt, 2019-05-01:
Hi nouyang!
I think the problem you have is because it miss the config files in 'https://github.com/NVIDIA/Dataset_Utilities/tree/master/nvdu/config/object_settings'.
In order to fix this you should clone the git repo then run `pip install -e .` inside the cloned folder (you can follow instructions here: https://github.com/NVIDIA/Dataset_Utilities#install-from-source-code-git-repo). That way the module will have reference to the config files. This is a problem with the package's `setup.py` and manifest files which cause it to not include the config file.
Please let me know if it fix your problem or not!

nouyang, 2019-05-02:
Hello thangt,
Yes, that does! Ah, I see the line now... got buried in the 3 readme's I had open to get nvidia-docker, dope-docker, and then dope running. I guess in retrospect it wasn't even needed just to get DOPE running on webcam images. 
Thanks!

closed 2019-05-07 by nouyang