NVlabs/Deep_Object_Pose #42: NDDS Question: Unreal engine interface angles to the rotation exported json

MohamadJaber1, 2019-05-16:
Hello,

First of all, thanks for sharing this amazing work!

Secondly, using **NDDS**, I would like to know the convention used to export the rotation matrix in the images _json_ files.

[image]

Here, the rotation angles in unreal engine interface are **[0, 0, -90]**. So I know that they have been exported in a way as OpenCV expects where each has a different coordinate system #28. For the translation vector, I knew how it has been ordered but for the rotation matrix being exported in the json file which is **[[-1 0 0][0 0-1][0 1 0]]**. So, I am trying to infer from this rotation matrix then angles to represent the same position of the object as shown above.  I tried calculating them but its not matching.

Thank you once again, looking forward for your reply.

TontonTremblay, 2019-05-16:
I dont remember exactly and the dev for NDDS is on vacation this week and the next one. Could you not just use the quaternion and then create a rotation matrix out of it to double check it?  

We also released a tool to visualize the data from NDDS, you can probably find your answer somewhere in there (https://github.com/NVIDIA/Dataset_Utilities). 

Just for fun, could you share the json file and the _camera_setting.json and the _object_setting.json?

MohamadJaber1, 2019-05-17:
Thank you for your reply @TontonTremblay

I will try looking into it. It caused me headache trying to figure out the convention used to export this rotation matrix. 

The rotation angles given at unreal engine interface are [0 0 90] degrees. 

 _Camera_setting.json
[image]
_object_setting.json
[image]#

000000.json (**note that the second column of the rotation matrix is different from that of the object settings one**)
[image]

Hope it is possible to infer the rotation angles from the above rotation matrix. Thanks.

MohamadJaber1, 2019-05-20:
Quaternions solved my problem, thanks a lot :smile:

closed 2019-05-20 by MohamadJaber1

SoumyaViralint96, 2021-01-18:
Hi, all i have a doubt can we extract all the features at a time for a dynamic object.

bcz when I try to extract all the data the textures information is missing on the meshes. 

please I need help!

Thanks.. 
[image]

thangt, 2021-01-21:
```when I try to extract all the data the textures information is missing on the meshes.```
What information is missing? Can you show us some images of this problem? 
Also can you create a new thread since your question doesn't seem to fit this issue.

SoumyaViralint96, 2021-01-28:
Hi, I am having issues with the exported images using NDDS. Upon exporting, the texture/color of the object is missing whenever I try to also export depth, segmentation masks, etc. The colors are only exported properly when the True color option is selected.
image
Here is a sample of the resulting image.
000037

When all data are selected the exported image becomes like this:
000009

Can you help me in fixing this problem?

Thanks.

SoumyaViralint96, 2021-01-28:
Hi, I am having issues with the exported images using NDDS. Upon exporting, the texture/color of the object is missing whenever I try to also export depth, segmentation masks, etc. The colors are only exported properly when the True color option is selected.
[image]

Here is a sample of the resulting image.

[image]

When all data are selected the exported image becomes like this:
[image]

Can you help me in fixing this problem?

Thanks.