NVlabs/Deep_Object_Pose #44: TypeError: cameraMatrix is not a numpy array, neither a scalar

Luedeke, 2019-05-22:
Hello together,
I tried to implement the Ros code in the train.py to get to the projected_points to calculate the metric. But I always get the following error (sometimes directly at the first call sometimes at a later call):

```
[ INFO:0] Initialize OpenCL runtime...
Traceback (most recent call last):
  File "test2.py", line 1944, in <module>
    _runnetwork(epoch, testingdata, train=False)
  File "test2.py", line 1821, in _runnetwork
    config_detect
  File "/home/nils/catkin_ws/src/dope/src/inference/detector.py", line 268, in detect_object_in_image_alreadytensor
    detected_objects = ObjectDetector.find_object_poses(vertex2, aff, pnp_solver, config)
  File "/home/nils/catkin_ws/src/dope/src/inference/detector.py", line 304, in find_object_poses
    location, quaternion, projected_points = pnp_solver.solve_pnp(points)
  File "/home/nils/catkin_ws/src/dope/src/inference/cuboid_pnp_solver.py", line 88, in solve_pnp
    flags=pnp_algorithm
TypeError: cameraMatrix is not a numpy array, neither a scalar

```

The variable cameraMatrix does not exist with me? But the matrix_camera variable exists.  I don't understand where the error comes from.

Also I don't understand where the info comes from: [ INFO:0] Initialize OpenCL runtime... comes from.

Here is my Code:
[training.tar.gz](https://github.com/NVlabs/Deep_Object_Pose/files/3207484/training.tar.gz)

Thanks for your help.

Luedeke, 2019-05-24:
Ah! The output is two-dimensional but for the calculation of the ADD-metric I need the three dimensional representation. For this I have to calculate the three dimensional metric using the quaternation and the location? Correct?

TontonTremblay, 2019-05-24:
If you read the paper you would notice that the neural network outputs belief maps to know where the cuboid object is located. Using that information and the fact that we know the size and have access to the object size and the camera intrinsic we find the 3d pose of the object using PnP.

closed 2019-05-24 by TontonTremblay

Luedeke, 2019-05-28:
Yeah, I got that, thanks! But in your implementation, the results are the 2D points. 
But for the ADD metric you need the predicted pose in 3D?
So I have to extend the PnP in cuboid_pnp_solver.py in method solve_pnp to 3D points?

mintar, 2019-05-28:
You don't need to change anything. The translation and rotation are in `result['location']` and `result['quaternion']` respectively, see here:

https://github.com/NVlabs/Deep_Object_Pose/blob/afea0462b831804ef84157631b3d379fd4a6a510/src/dope.py#L215-L216

Luedeke, 2019-05-28:
@mintar I'm afraid I'm lost on this one. Excuse me.
So to get the predicted pose, you calculate the location of the cuboid with the translation and rotation?

mintar, 2019-05-28:
A pose is a combination of a position (translation) and a rotation. The translation and rotation given in the two variables above *are* the predicted pose of the object. The meshes used by DOPE are translated such that the mesh origin is in the center of the 3D bounding box (= cuboid). That means that the pose of the object and the pose of the cuboid are the same. I hope that clarifies things.

mintar, 2019-05-28:
By the way, the poses returned by DOPE are in centimeters, while ROS uses meters, so you might have to divide the location by a factor of 100. Just look at the lines surrounding the code I linked to (around line 215); there, a ROS Pose message is built and published.

Luedeke, 2019-05-28:
@mintar Thank you, that was very helpful. 
But I want to calculate the ADD Metric between the ground truth and the predicted pose using [this](https://github.com/NVlabs/Deep_Object_Pose/issues/13#issuecomment-488727546) code . I needed this 4x4 matrix (?), as in the JSON file in the pose_transform_permuted variable or in newer pose_transform:

```
"location: [ 3.636699914932251, -7.1244997978210449, 61.205299377441406 ],
"quaternion_xyzw": [ 0.49849998950958252, -0.50099998712539673, -0.23849999904632568, .66600000858306885 ],
"pose_transform_permuted": [
	[ -0.90520000457763672, -0.42500001192092896, 0.0010000000474974513, 0 ],
	[ 0.38409999012947083, -0.81720000505447388, 0.42960000038146973, 0 ],
	[ 0.1817999929189682, -0.38929998874664307, -0.90299999713897705, 0 ],
	[ 3.636699914932251, -7.1244997978210449, 61.205299377441406, 1 ]
],
```

So I have to get out of this result:

```
[{'cuboid2d': 
array([None, None, None, None, None, (393.12350866760227, 8.2176658678339827),
       None, (328.01836364338948, 235.36015965875694),
       (368.21523734750838, 112.01849561320836),
       (334.27798525944007, 116.20128223083461)], dtype=object),
'name': 'cracker', 
'quaternion': Quaternion([ 0.48902345, -0.48971287, -0.3328639 , 0.64049902]), 
'location': [0.87687700075507136, -12.8709826562585, 69.360323550436604],
 'projected_points': array([[ 324.85022498, -15.03408885],
       [ 269.99226383, 130.28539506],
       [ 273.83800017, 212.40307047],
       [ 314.88919115, 98.09603198],
       [ 396.21567149, 6.20608956],
       [ 347.30944482, 154.16019378],
       [ 330.73077727, 230.28894467],
       [ 368.49273695, 114.47831118],
       [ 328.11006303, 120.95881116]])}]
```

a 4x4 matrix to compute the ADD error for a given cuboid. 
```
[[?, ?, ?, 0]
 [?, ?, ?, 0]
 [?, ?, ?, 0]
 [0.87687700075507136, -12.8709826562585, 69.360323550436604, 1]]
```

But I am not sure how to calculate the remaining parameters. I thought I'd calculate this 4x4 Matrix with the rotation and translation. Or is that wrong?

Luedeke, 2019-06-06:
@mintar thank you very much for your answers! The groundtruth pose from the JSON file from NDDS is also in centimeters?

mintar, 2019-06-06:
Yes, that's also in centimeters.

Luedeke, 2019-06-06:
@mintar Great. I have just tested the output of the cracker model with a single image from the FAT record. 
In the Groundtruth it says:
"location: [ -3.9193000793457031, 5.9113001823425293, 86.549697875976563 ],
"quaternion_xyzw": [ -0.027100000530481339, 0.90950000286102295, 0.21549999713897705, -0.35449999570846558 ],
"pose_transform_permuted": [
[ -0.65659999847412109, 0.37279999256134033, -0.65570002794265747, 0 ],
[ -0.74709999561309814, -0.20219999551773071, 0.63319998979568481, 0 ],
[ -0.10350000113248825, -0.90560001134872437, -0.41130000352859497, 0 ],
[ -3.9193000793457031, 5.9113001823425293, 86.549697875976563, 1 ]
]
But in the output of DOPE comes the following, although DOPE gives an almost perfect result. how can this be? 

location: [14.285128324083644, 9.4828578086467701, 73.954815002424127]
quaternion: [-0.01088066 -0.86830612 -0.18333475 0.46077593]
This would result in the following:
pose_transform:
[[ -0.79619952 0.32840847 -0.50814782 0. ]
 [ -0.57513431 -0.15005699 0.80417873 0. ]
 [ -0.18784797 -0.93253996 -0.30835428 0. ]
 [ 14.28512832 9.48285781 73.954815 1. ]]

Luedeke, 2019-06-12:
@TontonTremblay Is it possible to give detect.detect_object_in_image multiple tensor images? Directly from 
       ` data = Variable(targets['img'].cuda()) `

TontonTremblay, 2019-06-12:
I am not sure I understand the question, could you be more specific?

Luedeke, 2019-06-13:
@TontonTremblay  I want to use the images from the Dataset in train.py from:
    ` data = Variable(targets['img'].cuda())`
Now we have multiple tensor Images from the Dataset. And I want to check for each Image if there is a object to detect:
`results = ObjectDetector.detect_object_in_image_alreadytensor(
                        net,
                        pnp_solvers[model],
                        data,
                        config_detect
                    )`

I created this method, because the images are already tensors:
```
@staticmethod
    def detect_object_in_image_alreadytensor(net_model, pnp_solver, image_torch, config):
        '''Detect objects in a image using a specific trained network model'''

        if image_torch is None:
            return []

        # Run network inference
        out, seg = net_model(image_torch)
        vertex2 = out[-1][0]
        aff = seg[-1][0]

        # Find objects from network output
        detected_objects = ObjectDetector.find_object_poses(vertex2, aff, pnp_solver, config)

        return detected_objects
```

And im not sure if the detect_object_in_image_alreadytensor or the detect_object_in_image can handle multiple images at once? Or do I have to split this data tensor ` data = Variable(targets['img'].cuda())` into his images?  If I need to split data, how can I make multiple tensor images out of it? With` for img in Data` it doesn't seem to work.

Luedeke, 2019-06-13:
Another question, you're using Data Augmentation. So how can I use the rotation and translation Matrix from Reproject to pose_transform (ground truth). So that the Groundtruth is also rotated and transformed in the same way?

TontonTremblay, 2019-06-14:
You are correct in that `find_object_poses` will only work with single images. 

I will refer you to the dataloader around line 535 in how to use `reproject`.

Luedeke, 2019-06-14:
@TontonTremblay  Thank you so much. But this line, it outputs multiple tensor images, doesn't it? How can I divide them?
`data = Variable(targets['img'].cuda())`

 I thought with a For but that doesn't seem to be correct.
`for image in data:`