NVlabs/Deep_Object_Pose #50: Quaternions not changing

baljit92, 2019-07-01:
I've been generating custom data using the NDDS plugin. However, there is one parameter that I do not understand which is ```quaternion_xyzw``` under ```object_settings``` in the ```00.json``` file. The value for this field is always ```[0,0,0,1]``` for all the generated images. I have written blueprint that rotates the object continuously when I press the play button and frames are captured.

Why is that the ```quaternion_xyzw``` value not change with rotation?

thangt, 2019-07-01:
Do you have some example data for me to look at? It's better if you can include both images and the annotated data (.json).

baljit92, 2019-07-01:
I've uploaded the files. Image 000000.png is the starting position and image 000002.png is the one with the hand rotated. ```json.txt``` has the parameters for both images

[image]
![000002
[json.txt](https://github.com/NVlabs/Deep_Object_Pose/files/3347020/json.txt)
](https://user-images.githubusercontent.com/5225501/60460553-be8efc00-9c4c-11e9-8224-d8bfb60c11b6.png)

thangt, 2019-07-01:
The image and the projected_cuboid does look correct - you can use the nvdu tool to visualize it: https://github.com/NVIDIA/Dataset_Utilities#visualize-a-dataset-generated-by-ndds.
I notice some that you didn't set the class name of the hand. You can change it in the `Tag` property of the `AnnotationTag` component.
I can see that the `quaternion_xyzw` stay as [0, 0, 0, 1] and but the `pose_transform` matrix does seem to have the rotation updated. Which version of the NDDS you used to generate this data?

baljit92, 2019-07-01:
I'm using ```v1.2``` with Unreal Engine ```4.22.3```. So shouldn't the ```quaternion_xyzw``` array change when the object is rotated?

thangt, 2019-07-01:
Yes it should. You can look at the pose_transform matrix, it's the 4x4 transform matrix of the object which include the 3x3 rotation matrix and 3x1 translation (last row). I tried in my side and can see the quaternion_xyzw working as intended. 
How did you create the hand actor to be captured? Did you use the DRAnnotatedActorBP? Is the hand a StaticMesh or SkeletalMesh?
If you can actually send me the .uasset of the hand and its actor or even the .umap file, it will help me debug this issue even faster.

baljit92, 2019-07-01:
Here the gitlab link: https://gitlab.com/baljit_singh/hmcresearch
You can access the file under Data Collection/Unreal Projects/Source/Content/AutoRotation

Let me know if you need anything else

thangt, 2019-07-01:
The problem is caused because you used scale on the hand. I have an internal fix for it and we will push it in the next patch (should be soon).
In the mean time you can scale the mesh first before assigning it to the static mesh. Also I would recommend you to use DRAnnotatedActorBP (it's in /DomainRandomizationDNN/Blueprint/DR_AnnotatedActor_BP).  That blueprint will automatically set the tag name and center-align the mesh.

baljit92, 2019-07-01:
Got it! I didn't understand what you meant by ```scale the mesh first before assigning it to the static mesh```. How can I do this?

thangt, 2019-07-01:
When you import the 3d model of your object into UE4, it will ask you about scaling and additional 3d transforms to apply to the model. In your case, instead of scale 3 for the hand, you should put 30.
Since the main purpose of the NDDS is for pose detection of a real object, we want to make sure the size of the object stay the same, that's why we didn't pay a lot of attention for scaled objects. The next patch may make it a bit easier but I recommend you to try to measure the real object and try to make the 3d model in UE4 to be the same size (UE4 use centimeter units), it will make DOPE work better too since we use PNP to calculate the 3d pose from the 2d keypoints.

baljit92, 2019-07-01:
Understood. Let me try that. Similarly, if I need to set a certain position of the hand to be the starting point(origin); should I modify the rotation in the import dialog box itself or can I do that later after importing?

thangt, 2019-07-01:
Yes, you can modify the translation to change where the origin of the mesh is and change the import rotation to make the original pose of the hand point where you want it to.
For the origin location, you can use DR_AnnotatedActor_BP instead of just normal actor which will automatically calculate the center of the mesh and move mesh there to match it.

baljit92, 2019-07-01:
But what if I want my origin to be somewhere else and not in the center of the object mesh? For example; in the hand object; I want the center to be where the thumb touches the index finger.

thangt, 2019-07-02:
That's fine. You can turn off the flag `ForceCenterOfBoundingBoxAtRoot` of AnnotatedActor, it wont't automatically center-align.

baljit92, 2019-07-02:
It works. Thank you very much :)

closed 2019-07-02 by baljit92