NVlabs/Deep_Object_Pose #53: NDDS error on TestCapturer.umap

baljit92, 2019-07-11:
When I try to build and launch NDDS: I get an error saying "TestCapturer.umap is an asset" I've added a screenshot along. In addition to that, there is no SimpleScene_Capture under NVSceneCapturer Content.

Are these two issues related? Also, is this issue because of the new NDDS update? I'm using Unreal 4.22.3 on Fedora

[image]

thangt, 2019-07-11:
Can you check the TestCapturer.umap in the disk to see how big it is? This problem may happen if you don't use git LFS, which will cause LFS files to be just a text file.

baljit92, 2019-07-15:
Yes. That was the problem. Thank you!

closed 2019-07-15 by baljit92