NVlabs/Deep_Object_Pose #55: Why no images when I input rostopic echo /dope/rgb_points ？

Sunweijun333, 2019-07-16:


Sunweijun333, 2019-07-16:
/dope/webcam/image_raw has normal images,but no dectecting result.Thank you for your reply

TontonTremblay, 2019-07-16:
which object are you trying to detect? Are you sure the weights are not commented out in the confiig file?

Sunweijun333, 2019-07-16:
comment other five objects,and keep soup box,when I input rosrun dope dope ,rviz can display,but not real-time,no dectetct box.

Sunweijun333, 2019-07-16:
@TontonTremblay I use realsense435 camera

TontonTremblay, 2019-07-16:
Could you please share an example image where you can see the tomato soup can?

zhangstar3, 2019-12-27:
[image]
Oh,i also have the problem,i add  image in rviz,it shows a picture.and i add camera in rviz,and choose the topic ,but the is a warning say no image receive. i use the command rostopic echo /rgb_points returns so many numbers.
So ,where is the problem,and i want to show the picture in the camera module of rviz.

mintar, 2019-12-28:
@zhangstar3 : Your problem is different, the solution is in #70. In short: don't use the "Camera" display in RViz for the `rgb_points` topic. Either use the "Image" display for the `rbg_points` topic, or use the "Camera" display with your camera's `/camera/color/image_raw` (or similar) topic.

mintar, 2019-12-28:
I think this issue can be closed - no activity for 6 months.

zhangstar3, 2019-12-28:
@mintar Oh,thanks for you reply, if i not use the topic "rgb_point" in camera, it did shows the picture when i select the topic of "image_raw",But i see others video,he could use the camera in dispaly in Rviz,and he shows the xyz axis in the object.because i also want to display the orientation of the object,so i want to use the camera in rviz.
enen,whether i should publish topic about the pose and orientation of the object  by myself, if i want get the same of the screenshot shows ??
[image]

PR #121 (2020-09-22): Publish camera_info topic for rgb_points
...and also properly set the time stamp of rgb_points.

This allows visualizing the rgb_points using the Camera display in RViz.

See #119.

This fixes #55 and fixes #70.

Do not merge yet, I haven't been able to test it yet.
commit e8b58683 (2020-09-22): Publish camera_info topic for rgb_points

...and also properly set the time stamp of rgb_points.

This allows visualizing the rgb_points using the Camera display in RViz.

See #119.

This fixes #55 and fixes #70.
closed 2020-09-22 by TontonTremblay