NVlabs/Deep_Object_Pose #58: How to Set Custom Annotation Cuboid in NDDS?

Abdul-Mukit, 2019-07-25:
Hi. My research requires me to track the pose of a [pen](https://image.made-in-china.com/43f34j00MetEiVJmrwuI/Disposable-Cautery-Pen.jpg) (sort of a knife) while it is in a surgeon's hand. I have attached a sample image of my 3D model.
[image]

The problem is that NDDS draws the 3D bounding box around the entire mesh. However, I only need the bounding box around the pen. So I wanted to know if manually setting the 3D bounding box ranges is already a feature in NDDS and how to do it?
I did find the settings of NVAnnotated Actor as shown in the image. However, they can't be edited. If they could be edited, then maybe this problem would be solved.

If this feature doesn't exist, then I will attempt to make that feature for my research. I just needed to know from the authors if this feature can be added or if there are any technical limitations. Also, where should I get started with the coding?

A little background:
I already trained DOPE multiple times on dataset created using only the 3D model of the pen. However, [performance is not robust](https://youtu.be/g1adPMSmrXY); especially in extreme lighting of Surgical Lighting. We did a cadaver lab with OR-like environment. Surprisingly DOPE with the same weights as in the video couldn't detect anything. After our experience, we decided to add the hand model as the pen needs to be tracked only when the surgeon is about to use it.

Thank you, very much for your support.

thangt, 2019-07-25:
Do you have the 3d model of the pen and hand separatedly? If so, you can create another actor as the hand and attach the pen (AnnotatedActor) to it. That way you can randomize the color/texture of the hand, also since the pen is attached to the hand, when you rotate the hand the pen will also rotated.

Step by step of the setup:
- Create a blueprint of type AnnotatedActor for the pen - let's name it PenAnnotated_BP. I assume you already did this for your previous dataset.
- Create an StaticMeshActor for the hand - let's name it HandActor. I suggest you create a blueprint for this, but for convenience, you can just drag the hand's mesh into the level.
- Add ChildActorComponent to the HandActor - let's name this component PenChildComp. In ChildActorComponent, in the field ChildActorClass, select PenAnnotated_BP. Note that the pose of the PenChildComp is the pose of the pen. You can move and rotate PenChildComp to get what you want.

About your test environment, I think you should increase the number of light source and increase the randomizable intensity range of those lights. Also the material of the pen with how accurate its reflection  compare to the real life help a lot too.

Please let me know if this setup help you or not.

Abdul-Mukit, 2019-07-25:
@thangt Thank you, very much. It worked. 
I created the blueprint HandActor just as you said. One thing I did, in addition, was that I set the mobility of the Box-Component for the PenAnnotated_BP to static. Which made the pen and the hand model both stationary when I ran NDDS. I don't know if that is the right way to do it but it did solve my problem. 

Regarding the suggestions about the testing environment and materials, I'll try those out too. Thank you again.

closed 2019-07-26 by thangt

akeaveny, 2020-06-20:
Just as a follow up to @Abdul-Mukit's comments, can I ask:

1. How is the bounding volume set when importing object models (e.g. from Blender)? Is it minimum volume to contain the object? 

2. Does nvdu_ycb align the cordinate axes to the ~center of mass of the object?

3.  Also, for the FAT dataset why is it that we want to detect pose for the entire object? If pose is detected to the centroid of the bounding volume wouldn't it be possible that the centroid does lie on the object? This makes sense for simple objects like the soup can but not for the powerdrill.

[image]

I've included a snip for a hammer object where I'd like to detect pose for the handle.

[image]

Thanks!

Aidan

TontonTremblay, 2020-06-23:
Hello, 

1) We ended up measuring the objects we had in the robot room, some of them had up to 7mm difference with the 3d models. For 3d exporting we just used the aabb from the 3d models. 
2) @thangt can you answer that question?
3) Follow up work has focused on finding keypoints on the actual object (pvnet). There is nothing stopping you in using the bounding box you defined there. @thangt could you help him how he can do it in NDDS?

akeaveny, 2020-08-17:
Apologies for not getting back til now.

I manged to get all my child cuboids for the YCB objects. One thing I haven't been able to solve, similar to @KatharinaSchmidt  in https://github.com/NVIDIA/Dataset_Synthesizer/issues/72, is randomly spawing object with the correct segmentation ID. The rgb and depth images are ok.

I've included a picture for my '001_master_chef_can_03' which should have an ID of 1 using the actor tag in my scenemanager. I've verified that the tag is ok if I manully drop the actor in the scene rather than using the group manager. The segmentation ID is 32, not 1, with the group manager (I've a total of 31 tags)

[image]
[image]