NVlabs/Deep_Object_Pose #59: How to use SceneMarker_Volume or SceneMarker_FocalPoint in NDDS?

Abdul-Mukit, 2019-07-25:
Hi. My objective is to rotate the camera around my target object from a range of 60-80cm and keeping a positive azimuth of 30-90 degrees. I have gone through the documentation of NDDS but nothing is mentioned about how to use them. It would be very helpful if the instructions of their usage are given.

Thank you.

thangt, 2019-07-25:
You can add the OrbitalMovementComponent to your camera. You can mouse over each of the component's properties to read the tooltips, some of the main properties:
- FocalTargetActor - you need to select an actor in the level as the focal target
- PitchRotationRange - this is the attitude range
- YawRotationRange - this is the azimuth range
- TargetDistanceRange - how far the camera stay from the focal target, 60-80cm
  - NOTE: You need to turn on ShouldChangeDistance flag first
- Other properties allow you to change the speed of the rotation and distance change

Please let me know if it work out for you or not.

Abdul-Mukit, 2019-07-28:
Thank you, again. It [worked](https://youtu.be/2mU7I_q4OHg) fine.

closed 2019-07-28 by Abdul-Mukit