NVlabs/Deep_Object_Pose #64: NVCapturable Actor Tag not working when used in a blueprint

Abdul-Mukit, 2019-08-12:
Hi,
Thank you again for such a great repository. I have been having some problems with the NVCapturable Actor Tag. When it is used on a static mesh as shown in the Tutorial-1 video, it works well. However, when the Actor Tag component is added to a blueprint it doesn't seem to work. In both cases, the "Class Segmentation Type" was set to "Actor Tag" in the SceneManger_BP (tried out other settings as well). 

For example, the Test_AnnotatedActor_BP4 in the "Test Capturer" level has the "Annotation Tag" component which doesn't work. The output class is always the Mesh-Name of whichever mesh I select. I tried to create a new child blueprint from the DR_AnnotatedActor_BP too but still the "Annotation Tag" isn't working.

Can you please let me know if I am doing anything wrong here?

My application requires me to randomize mesh (hands) but all objects will have the same class name (hand). I made a BP out of the DR_AnnotatedActor_BP and added the RandomMesh component to it. However, now that the AnnotationTag isn't working I am stuck. Would really appreciate if you could suggest a solution.

p.s. I am on Linux using the latest DOPE.

thangt, 2019-08-12:
In the Test_AnnotatedActor_BP, the blueprint type itself have property 'Set Class Name from Mesh' in the "Annotated Actor" category (this is the actor property, not its child component). If you set it to true, the AnnotationTag's name will be overwritten to be the mesh's name. It's by default set to true, you can set it to false and try again.

Abdul-Mukit, 2019-08-12:
Worked. Thank you very much. Will go through all the properties category from now.

closed 2019-08-12 by Abdul-Mukit