NVlabs/Deep_Object_Pose #65: Determining pose of glass objects

dj4park, 2019-08-14:
Has there been any experiment done with using DOPE to determine the pose of a glass bottle? If not, would there be any reason as to why it would not work?

Currently I am experimenting with a Soju bottle. I have trained the network using 6000 basic NDDS images for 60 epochs using a batch size of 4 and a learning rate of 0.001 but I am having difficulties in detection. 

Could the problem be due to the hyperparameters, image quantity/quality? Or could it be due to the texture of the glass bottle?

TontonTremblay, 2019-08-15:
Do you have symmetries on you object, could you show me some training examples and testing examples? I never tried transparent objects with DOPE.

dj4park, 2019-08-15:
Following a similar strategy to @Abdul-Mukit and his [video](https://www.youtube.com/watch?v=g1adPMSmrXY), I created 6000 simple images using NDDS and trained the model for 60 epochs on a NVIDIA Quadro P3200 Mobile setting the learning rate to 0.001 and the batch size to 4.

Here is a photo of the object that I created using Blender.
[image]

And here are some training examples that I made with NDDS.
[image]
[image]
[image]
[image]

And here is the actual object itself with which I am testing the model.
[image]
[image]

Abdul-Mukit, 2019-08-15:
The contributors/authors will give the best suggestions, but as I am working on things similar to this I would like to share some opinion. Your 3D model doesn't look as transparent as the original bottle, the labels don't match either. The smoothness of the surface is off I think. I think you should edit the material, and increase metalic property and decrease roughness. You should also work on making the model more transparent. The cap color doesn't match either. The label on the neck of the bottle is missing in your model. The cap doesn't have the same material as the glass and the mesh of the cap doesn't look like the original cap. For the labels of the bottle, you can scan the label image and use it as a texture for the 3D model. You may use u-v mapping to do that - not sure if that will work but I would have tried that. Also, try randomizing the background with images. In my small experience, I have felt background randomization with real images tends to prevent overfitting better compared to the small number of abstract background generated by NDDS.

TontonTremblay, 2019-08-16:
I second!

aditya2592, 2019-08-26:
Hi @dj4park. Were you able to successfully train for your custom object?

dj4park, 2019-08-29:
I am randomizing the background with the default images provided with NDDS.

sanjaysswami, 2019-10-15:
> Following a similar strategy to @Abdul-Mukit and his [video](https://www.youtube.com/watch?v=g1adPMSmrXY), I created 6000 simple images using NDDS and trained the model for 60 epochs on a NVIDIA [...]

How did you insert that exact texture (green color and written things etc) in blender? Rather can you please explain me how did you create this 3d model? I a also trying to generate the dataset for my own object but I don't know how to add textures. If you can help me with this issue it would be really greate help to me.

closed 2025-08-08 by nv-jeff