NVlabs/Deep_Object_Pose #68: What are “belief maps” and “affinity maps”?

hamza-04, 2019-08-21:
i've tried to look for these termes, “belief maps” and “affinity maps”, but after spending a good time in research, these words still confusing for me , and i can't explain them as well . 
Can you give a clear explanation to them . 

by the way, i found pictures  shared in the issues of belief maps, but not understanding them ?
thank you in advance,

TontonTremblay, 2019-08-21:
Think of belief map as a 2d heatmap of where a keypoint is located, e.g., high value is equivalent to high likelyhood of finding a point here. When you take the maximum locations on that map then you find where the keypoint is located. Now because DOPE can deal with multiple instances of a single object, we need to be able to dissociate the different objects and thus each keypoint map is also associated to an affinity field, think about it as a vector field. Each pixel outputs a vector direction x,y. That vector is normalized and it is pointing towards somewhere.

hamza-04, 2019-08-22:
thank you so much for your answer . I want to ask another question, the threshold who keep only the high values of the belief map or "HEATMAP", this threshold how it is defined or used ? 
And the "HEATMAP", how do we get it ? like how we can give a special pixel of the imput image a certain heat, and consequently the vertices have, fortunately, a higher heats (for sure it depends on the network). To sum up, i want to know the genius behind predicting nine "heats" or peaks from an imput rgb image ?       

thank you in advance for your response , and your contribution of work 

Hamza

Richo400, 2021-05-17:
Hi, awesome job with DOPE , I find myself testing your work and have found the pose with the ROS inference with the box of cookies. I would like to see the belief and affinity maps, in what way do I display those screens. Thank you very much

closed 2025-08-08 by nv-jeff