NVlabs/Deep_Object_Pose #70: Lost some topic after start the dope.launch

daniellee8527, 2019-08-26:
I followed the tutorial , but I encountered some problem about rostopic and rviz.
First, I started  `roscore`

Second, I try to start my own camera(Intel RealSense D435i) node , the following is my camera.launch : 
      `roslaunch realsense2_camera rs_camera.launch`
After I start this node, I type `rostopic list` to check, here is some topic listed : 
`/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
`

Third, I followed the 3rd step on github :Edit config info (if desired) in `~/catkin_ws/src/dope/config/config_pose.yaml`
So I edited it, changing the `topic_camera` and `topic_camera_info` from default into the topic published by my own camera node, here is my change in `config_pose.yaml`:
`topic_camera: "/camera/color/image_raw"`
`topic_camera_info: "/camera/color/camera_info"`
`topic_publishing: "dope"`
`input_is_rectified: True`
`downscale_height: 500`
`frame_id: "/camera_link"`

Forth, after edited the `config_pose.yaml`, I started the dope node by `following:roslaunch dope dope.launch [config:=/path/to/my_config.yaml]`

Now, I type `rostopic list` again, I saw some topic about dope,
`/dope/detected_objects`
`/dope/dimension_cracker`
`/dope/markers`
`/dope/pose_cracker`
`/dope/rgb_points`

But I didn't see the `/dope/webcam_rgb_raw` , I open the `Rviz`, Add `Image`, `Pose`, `MarkerArray`, and `camera` , but only `camera ` not working, the cameara status is : `No image received.` . Here is the screenshot of my rviz : 
[image]

Is this condition normal ?

mintar, 2019-08-26:
You seem to be using an outdated version of dope. In the newest version, there is no "webcam_rgb_raw" topic or "frame_id" parameter. Could you please update and try again?

daniellee8527, 2019-08-26:
Thanks for your quick answer .
But I still have a question : where can I get the newest version ? 
Because I followed this Github website : `https://github.com/NVlabs/Deep_Object_Pose` today , it means that I git clone this open source today, so my version should be the newest ?
Or, does any new github website talks about the new version ?

mintar, 2019-08-26:
Ok, then you have the correct one. Probably your `config_pose.yaml` file is still outdated. Could you use the newest `config_pose.yaml`, and then only change the topic names?

daniellee8527, 2019-08-26:
Thanks for your quick answer .
I git clone all resource today ,  the `config_pose.yaml` also included in, so I think this version should be the newest one. I only changed two part in the default setting : `    
topic_camera: "/dope/webcam/image_raw"`and `topic_camera_info: "/dope/webcam/camera_info"` 
to my camera topic`topic_camera: "/camera/color/image_raw"` and
`topic_camera_info: "/camera/color/camera_info"`,.
I just commented the `frame_id` line minutes ago , and tried again,  but in Rviz, the camera still not receive the image; by the way, others like `Image`,`Pose`,`MarkerArray` still worked normally.

mintar, 2019-08-26:
Oh, I see. I missed completely that the RViz *image* display for the topic `/dope/rgb_points` is working, but the *camera* display is not. That means that your dope config should be okay, at least the node is receiving stuff and publishing the right output image. I'll try to reproduce this when I find time.

mintar, 2019-08-26:
One thing you could try: In RViz, under "Global Options", set the "Fixed Frame" to the frame from the camera topic. You can find out what that frame is by running `rostopic echo -n1 --noarr /camera/color/image_raw | grep frame_id`.

daniellee8527, 2019-08-27:
Thanks for your answer. 
After I entered `rostopic echo -n1 --noarr /camera/color/image_raw | grep frame_id`, it showed `frame_id: "camera_color_optical_frame"`, and I set it into `config_pose.yaml` , then start dope node , but the condition is same to mentioned above.

Rviz -- Image can show the correct ROI of object , I also found that `Pose` worked , but the arrow of X-Y-Z coordinate didn't show , `MarkerArray` status show `No mesh....` , is it due to the YCB models not installed ?

mintar, 2019-08-28:
I just tried it, and remembered everything. So:

* The RViz "camera" display does not work with the `/dope/rgb_points` topic. This is because dope does not publish the `/dope/camera_info` topic. The reason is that `/dope/rgb_points` is just for some quick debugging; you don't really need it. In my opinion, it is much better to use your input camera image for that, and also enable the dope markers. In short: **Use the topic `/camera/color/image_raw` for the "camera" display, if that is what you use for the input to DOPE.**
* You should not use the "raw" camera image for DOPE anyway. Please use a rectified image instead.
* Regarding the Jetson - you should get 1-2 frames per second: https://github.com/NVlabs/Deep_Object_Pose/issues/2#issuecomment-513935839

daniellee8527, 2019-09-02:
Thank for your opinion , I tried it , and the marker showed on "camera" display. 
Thank to help me resolve all of my problem.

mintar, 2019-09-02:
Glad that I could help you! :)

@TontonTremblay : This ticket can be closed now.

PR #121 (2020-09-22): Publish camera_info topic for rgb_points
...and also properly set the time stamp of rgb_points.

This allows visualizing the rgb_points using the Camera display in RViz.

See #119.

This fixes #55 and fixes #70.

Do not merge yet, I haven't been able to test it yet.
commit e8b58683 (2020-09-22): Publish camera_info topic for rgb_points

...and also properly set the time stamp of rgb_points.

This allows visualizing the rgb_points using the Camera display in RViz.

See #119.

This fixes #55 and fixes #70.
closed 2019-09-04 by daniellee8527