NVlabs/Deep_Object_Pose #73: off-by-one in CreateBeliefMap

RapIt, 2019-09-07:
Not that it makes any considerable difference (more of a note than an issue), but I think there is a off-by-one in the CreateBeliefMap function in the training code:
[https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train.py#L850#L851](url)
It is
```
for i in range(int(p[0]) - w, int(p[0]) + w):
          for j in range(int(p[1]) - w, int(p[1]) + w):
```
but in my understanding it should be ` int(p[0]) + w + 1`, otherwise the generated target belief-map points are slightly asymmetric.
Maybe this is also changes the OFFSET_DUE_TO_UPSAMPLING in the detector.

TontonTremblay, 2019-09-08:
You are correct this is a mistake, and I have made that change on my internal version. But this does not impact the location of the keypoint, it only affects how the peak looks like, e.g., it is missing a row and column of very low pixel values. If you want the github updated, please send a PR.

RapIt, 2019-09-08:
You are right, it does not change the center of the belief keypoint. No need to change the code, it doesn't make a difference anyway.

closed 2019-09-08 by RapIt