NVlabs/Deep_Object_Pose #74: why the loss goes to inf and nan?

qingchenkanlu, 2019-09-08:
yusheng@yusheng:~/code/Deep_Object_Pose-master/scripts$ python train.py --data /home/yusheng/fat/single/004_sugar_box_16k  --object 004_sugar_box_16k  --outf /home/yusheng/code --gpuids 0 1 --loginterval 1 --batchsize 32 --epochs 60
start: 17:12:18.049454
load data
training data: 188 batches
load models
Training network pretrained on imagenet.
Train Epoch: 1 [0/6000 (0%)]	Loss: 0.037806294858456
Train Epoch: 1 [32/6000 (1%)]	Loss: 0.037991054356098
Train Epoch: 1 [64/6000 (1%)]	Loss: 0.040447328239679
Train Epoch: 1 [96/6000 (2%)]	Loss: 0.084592938423157
Train Epoch: 1 [128/6000 (2%)]	Loss: 0.829165458679199
Train Epoch: 1 [160/6000 (3%)]	Loss: 9.005415916442871
Train Epoch: 1 [192/6000 (3%)]	Loss: 90.017181396484375
Train Epoch: 1 [224/6000 (4%)]	Loss: 1015.848876953125000
Train Epoch: 1 [256/6000 (4%)]	Loss: 15751.037109375000000
Train Epoch: 1 [288/6000 (5%)]	Loss: 354724.843750000000000
Train Epoch: 1 [320/6000 (5%)]	Loss: 14932532.000000000000000
Train Epoch: 1 [352/6000 (6%)]	Loss: 998436224.000000000000000
Train Epoch: 1 [384/6000 (6%)]	Loss: 83260162048.000000000000000
Train Epoch: 1 [416/6000 (7%)]	Loss: 7205331927040.000000000000000
Train Epoch: 1 [448/6000 (7%)]	Loss: 582404947312640.000000000000000
Train Epoch: 1 [480/6000 (8%)]	Loss: 41636405104869376.000000000000000
Train Epoch: 1 [512/6000 (9%)]	Loss: 2673606558960582656.000000000000000
Train Epoch: 1 [544/6000 (9%)]	Loss: 149008032411645116416.000000000000000
Train Epoch: 1 [576/6000 (10%)]	Loss: 7348256673702516621312.000000000000000
Train Epoch: 1 [608/6000 (10%)]	Loss: 314164264533932197806080.000000000000000
Train Epoch: 1 [640/6000 (11%)]	Loss: 5364948436383775982616576.000000000000000
Train Epoch: 1 [672/6000 (11%)]	Loss: 13268717647946179126755328.000000000000000
Train Epoch: 1 [704/6000 (12%)]	Loss: 28394457492577652746223616.000000000000000
Train Epoch: 1 [736/6000 (12%)]	Loss: 52918031761984019471794176.000000000000000
Train Epoch: 1 [768/6000 (13%)]	Loss: 106035777198682055404158976.000000000000000
Train Epoch: 1 [800/6000 (13%)]	Loss: 218677813065505417690349568.000000000000000
Train Epoch: 1 [832/6000 (14%)]	Loss: 450564491011966854653542400.000000000000000
Train Epoch: 1 [864/6000 (14%)]	Loss: 912350800526757883098955776.000000000000000
Train Epoch: 1 [896/6000 (15%)]	Loss: 1789880493922313488832659456.000000000000000
Train Epoch: 1 [928/6000 (15%)]	Loss: 3429768003024100588976603136.000000000000000
Train Epoch: 1 [960/6000 (16%)]	Loss: 6433352465997991795413221376.000000000000000
Train Epoch: 1 [992/6000 (16%)]	Loss: 11837126327259398519457316864.000000000000000
Train Epoch: 1 [1024/6000 (17%)]	Loss: 21323692376487189698974842880.000000000000000
Train Epoch: 1 [1056/6000 (18%)]	Loss: 36813930439282198319084863488.000000000000000
Train Epoch: 1 [1088/6000 (18%)]	Loss: 61357325200240187820289818624.000000000000000
Train Epoch: 1 [1120/6000 (19%)]	Loss: 98544587044192128811388633088.000000000000000
Train Epoch: 1 [1152/6000 (19%)]	Loss: 147204474636056691886480424960.000000000000000
Train Epoch: 1 [1184/6000 (20%)]	Loss: 213704367211178748821226127360.000000000000000
Train Epoch: 1 [1216/6000 (20%)]	Loss: 301466790433285109244961488896.000000000000000
Train Epoch: 1 [1248/6000 (21%)]	Loss: 418033136901927418820988764160.000000000000000
Train Epoch: 1 [1280/6000 (21%)]	Loss: 571614241889268924118103752704.000000000000000
Train Epoch: 1 [1312/6000 (22%)]	Loss: 770483937036354352772252958720.000000000000000
Train Epoch: 1 [1344/6000 (22%)]	Loss: 1022732051696589899563276435456.000000000000000
Train Epoch: 1 [1376/6000 (23%)]	Loss: 1344794423513750666564406870016.000000000000000
Train Epoch: 1 [1408/6000 (23%)]	Loss: 1744524990190690544037769773056.000000000000000
Train Epoch: 1 [1440/6000 (24%)]	Loss: 2212202325864705943303612268544.000000000000000
Train Epoch: 1 [1472/6000 (24%)]	Loss: 2717779485983639967196493381632.000000000000000
Train Epoch: 1 [1504/6000 (25%)]	Loss: 3234325381179664973420130467840.000000000000000
Train Epoch: 1 [1536/6000 (26%)]	Loss: 3784757164868033131753487990784.000000000000000
Train Epoch: 1 [1568/6000 (26%)]	Loss: 4357510554889765803168620347392.000000000000000
Train Epoch: 1 [1600/6000 (27%)]	Loss: 4905725168624960731595566743552.000000000000000
Train Epoch: 1 [1632/6000 (27%)]	Loss: 5421619452804213452694037200896.000000000000000
Train Epoch: 1 [1664/6000 (28%)]	Loss: 5920558250131916095959959863296.000000000000000
Train Epoch: 1 [1696/6000 (28%)]	Loss: 6413529185151091122304945487872.000000000000000
Train Epoch: 1 [1728/6000 (29%)]	Loss: 6903422799496437110085716541440.000000000000000
Train Epoch: 1 [1760/6000 (29%)]	Loss: 7399687452911152193617189142528.000000000000000
Train Epoch: 1 [1792/6000 (30%)]	Loss: 7913650780325025448266360160256.000000000000000
Train Epoch: 1 [1824/6000 (30%)]	Loss: 8448996983173332456443146665984.000000000000000
Train Epoch: 1 [1856/6000 (31%)]	Loss: 9004077086638118863953249435648.000000000000000
Train Epoch: 1 [1888/6000 (31%)]	Loss: 9591167732417571230065809686528.000000000000000
Train Epoch: 1 [1920/6000 (32%)]	Loss: 10178582370316680316797191192576.000000000000000
Train Epoch: 1 [1952/6000 (32%)]	Loss: 10785870539764434292531567525888.000000000000000
Train Epoch: 1 [1984/6000 (33%)]	Loss: 11417378329082347749152007389184.000000000000000
Train Epoch: 1 [2016/6000 (34%)]	Loss: 12066216070024436914991860285440.000000000000000
[... 254 lines cut ...]
    beliefsImg.append(Image.fromarray((stack*255).astype('uint8')))
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 2666, in fromarray
Process Process-11:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python2.7/dist-packages/torch/utils/data/dataloader.py", line 57, in _worker_loop
    obj = obj.tobytes()
KeyboardInterrupt
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "train.py", line 627, in __getitem__
    sigma = self.sigma)
  File "train.py", line 856, in CreateBeliefMap
    beliefsImg.append(Image.fromarray((stack*255).astype('uint8')))
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 2666, in fromarray
    obj = obj.tobytes()
KeyboardInterrupt
Traceback (most recent call last):
  File "train.py", line 1392, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1338, in _runnetwork
    target_belief = Variable(targets['beliefs'].cuda())        
KeyboardInterrupt

TontonTremblay, 2019-09-08:
Please give it a lower learning rate. I think this might be causing some exploding gradient. I never trained with such a small learning rate though.

qingchenkanlu, 2019-09-09:
@TontonTremblay Thanks! I have a try,but the result is awful.
if I set the batchsize=16, no matter what the lr is , I will get
Train Epoch: 1 [0/6000 (0%)]	Loss: 0.039806149899960
Train Epoch: 1 [16/6000 (0%)]	Loss: 0.040904726833105
Train Epoch: 1 [32/6000 (1%)]	Loss: nan
Train Epoch: 1 [48/6000 (1%)]	Loss: nan
Train Epoch: 1 [64/6000 (1%)]	Loss: nan
Train Epoch: 1 [80/6000 (1%)]	Loss: nan
Train Epoch: 1 [96/6000 (2%)]	Loss: nan

if I set the batchsize=32, lr=0.00001, I will get
yusheng@yusheng:~/code/Deep_Object_Pose-master/scripts$ python train.py --d/home/yusheng/fat/single/004_sugar_box_16k  --object 004_sugar_box_16k  --outf /home/yusheng/code --gpuids 0 1 --loginterval 1 --batchsize 32 --epochs 100 --lr 0.00001
start: 10:04:11.790992
load data
training data: 188 batches
load models
Training network pretrained on imagenet.
Train Epoch: 1 [0/6000 (0%)]	Loss: 0.043270923197269
Train Epoch: 1 [32/6000 (1%)]	Loss: 0.044183619320393
Train Epoch: 1 [64/6000 (1%)]	Loss: 0.044332496821880
Train Epoch: 1 [96/6000 (2%)]	Loss: 0.043765094131231
Train Epoch: 1 [128/6000 (2%)]	Loss: 0.044279184192419
Train Epoch: 1 [160/6000 (3%)]	Loss: 0.043717309832573
Train Epoch: 1 [192/6000 (3%)]	Loss: 0.042907532304525
Train Epoch: 1 [224/6000 (4%)]	Loss: 0.043558660894632
Train Epoch: 1 [256/6000 (4%)]	Loss: 0.043533191084862
Train Epoch: 1 [288/6000 (5%)]	Loss: 0.043324362486601
Train Epoch: 1 [320/6000 (5%)]	Loss: 0.042893439531326
Train Epoch: 1 [352/6000 (6%)]	Loss: 0.042370237410069
Train Epoch: 1 [384/6000 (6%)]	Loss: 0.043336097151041
Train Epoch: 1 [416/6000 (7%)]	Loss: 0.043230384588242
Train Epoch: 1 [448/6000 (7%)]	Loss: 0.043135456740856
Train Epoch: 1 [480/6000 (8%)]	Loss: 0.043726902455091
Train Epoch: 1 [512/6000 (9%)]	Loss: 0.044741559773684
Train Epoch: 1 [544/6000 (9%)]	Loss: 0.046717464923859
Train Epoch: 1 [576/6000 (10%)]	Loss: 0.049565110355616
Train Epoch: 1 [608/6000 (10%)]	Loss: 0.053250525146723
Train Epoch: 1 [640/6000 (11%)]	Loss: 0.059633512049913
Train Epoch: 1 [672/6000 (11%)]	Loss: 0.070787280797958

the loss becomes more and more greater

if I set the batchsize=32, lr=0.000001, I will get

yusheng@yusheng:~/code/Deep_Object_Pose-master/scripts$ python train.py --data /home/yusheng/fat/single/004_sugar_box_16k  --object 004_sugar_box_16k  --outf /home/yusheng/code --gpuids 0 1 --loginterval 1 --batchsize 32 --epochs 100 --lr 0.0000001
start: 09:48:32.386642
load data
training data: 188 batches
load models
Training network pretrained on imagenet.
Train Epoch: 1 [0/6000 (0%)]	Loss: 0.039792459458113
Train Epoch: 1 [32/6000 (1%)]	Loss: 0.040118962526321
Train Epoch: 1 [64/6000 (1%)]	Loss: 0.039407569915056
Train Epoch: 1 [96/6000 (2%)]	Loss: 0.039502039551735
Train Epoch: 1 [128/6000 (2%)]	Loss: 0.039573844522238
Train Epoch: 1 [160/6000 (3%)]	Loss: 0.039441470056772
Train Epoch: 1 [192/6000 (3%)]	Loss: 0.039882700890303
Train Epoch: 1 [224/6000 (4%)]	Loss: 0.039879325777292
Train Epoch: 1 [256/6000 (4%)]	Loss: 0.039967175573111
Train Epoch: 1 [288/6000 (5%)]	Loss: 0.039692126214504
Train Epoch: 1 [320/6000 (5%)]	Loss: 0.039724811911583
Train Epoch: 1 [352/6000 (6%)]	Loss: 0.039985574781895
Train Epoch: 1 [384/6000 (6%)]	Loss: 0.039775252342224
Train Epoch: 1 [416/6000 (7%)]	Loss: 0.039971295744181
Train Epoch: 1 [448/6000 (7%)]	Loss: 0.039414189755917
Train Epoch: 1 [480/6000 (8%)]	Loss: 0.038949500769377
Train Epoch: 1 [512/6000 (9%)]	Loss: 0.039555706083775
Train Epoch: 1 [544/6000 (9%)]	Loss: 0.040244515985250
Train Epoch: 1 [576/6000 (10%)]	Loss: 0.039147015661001
Train Epoch: 1 [608/6000 (10%)]	Loss: 0.039384994655848
[... 206 lines cut ...]
the loss seems doesn't converge or the converge is too slow. I only have train one epoch ,is the training time is not enough?

if I set batchsize=64, I will get 
start: 10:16:33.973653
load data
training data: 94 batches
load models
Training network pretrained on imagenet.
THCudaCheck FAIL file=/pytorch/aten/src/THC/generic/THCStorage.cu line=58 error=2 : out of memory
Traceback (most recent call last):
  File "train.py", line 1392, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1334, in _runnetwork
    output_belief, output_affinities = net(data)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 491, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/parallel/data_parallel.py", line 114, in forward
    outputs = self.parallel_apply(replicas, inputs, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/parallel/data_parallel.py", line 124, in parallel_apply
    return parallel_apply(replicas, inputs, kwargs, self.device_ids[:len(replicas)])
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/parallel/parallel_apply.py", line 65, in parallel_apply
    raise output
RuntimeError: cuda runtime error (2) : out of memory at /pytorch/aten/src/THC/generic/THCStorage.cu:58

others seems get same problem. I have two nvidia gtx 1080ti 11GB

TontonTremblay, 2019-09-09:
Can you remove the --object argument? I normally pass just sugar to train on it, but since you are training on single dataset, this will make sure it loads all the data.

qingchenkanlu, 2019-09-09:
It does work , the loss finally become more and more great T_T

yusheng@yusheng:~/code/Deep_Object_Pose-master/scripts$ python train.py --data /home/yusheng/fat/single/004_sugar_box_16k   --outf /home/yusheng/code --gpuids 0 1 --loginterval 1 --batchsize 32 --epochs 100 --lr 0.000001
start: 10:58:18.842664
load data
training data: 188 batches
load models
Training network pretrained on imagenet.
Train Epoch: 1 [0/6000 (0%)]	Loss: 0.038804907351732
Train Epoch: 1 [32/6000 (1%)]	Loss: 0.039279893040657
Train Epoch: 1 [64/6000 (1%)]	Loss: 0.038769133388996
Train Epoch: 1 [96/6000 (2%)]	Loss: 0.039265148341656
Train Epoch: 1 [128/6000 (2%)]	Loss: 0.038670811802149
Train Epoch: 1 [160/6000 (3%)]	Loss: 0.039157230407000
Train Epoch: 1 [192/6000 (3%)]	Loss: 0.038940127938986
Train Epoch: 1 [224/6000 (4%)]	Loss: 0.038488056510687
Train Epoch: 1 [256/6000 (4%)]	Loss: 0.038359414786100
Train Epoch: 1 [288/6000 (5%)]	Loss: 0.038298018276691
Train Epoch: 1 [320/6000 (5%)]	Loss: 0.038498181849718
Train Epoch: 1 [352/6000 (6%)]	Loss: 0.038484472781420
Train Epoch: 1 [384/6000 (6%)]	Loss: 0.038467951118946
Train Epoch: 1 [416/6000 (7%)]	Loss: 0.039112605154514
Train Epoch: 1 [448/6000 (7%)]	Loss: 0.038414534181356
Train Epoch: 1 [480/6000 (8%)]	Loss: 0.038314715027809
Train Epoch: 1 [512/6000 (9%)]	Loss: 0.038475196808577
Train Epoch: 1 [544/6000 (9%)]	Loss: 0.038775559514761
Train Epoch: 1 [576/6000 (10%)]	Loss: 0.038869984447956
Train Epoch: 1 [608/6000 (10%)]	Loss: 0.038977708667517
Train Epoch: 1 [640/6000 (11%)]	Loss: 0.038242880254984
Train Epoch: 1 [672/6000 (11%)]	Loss: 0.039328057318926
Train Epoch: 1 [704/6000 (12%)]	Loss: 0.038632005453110
Train Epoch: 1 [736/6000 (12%)]	Loss: 0.038621481508017
Train Epoch: 1 [768/6000 (13%)]	Loss: 0.039162911474705
Train Epoch: 1 [800/6000 (13%)]	Loss: 0.038663722574711
Train Epoch: 1 [832/6000 (14%)]	Loss: 0.038170833140612
Train Epoch: 1 [864/6000 (14%)]	Loss: 0.038872286677361
Train Epoch: 1 [896/6000 (15%)]	Loss: 0.038306020200253
Train Epoch: 1 [928/6000 (15%)]	Loss: 0.038344457745552
Train Epoch: 1 [960/6000 (16%)]	Loss: 0.038256756961346
Train Epoch: 1 [992/6000 (16%)]	Loss: 0.038965933024883
Train Epoch: 1 [1024/6000 (17%)]	Loss: 0.038473363965750
Train Epoch: 1 [1056/6000 (18%)]	Loss: 0.039064638316631
Train Epoch: 1 [1088/6000 (18%)]	Loss: 0.038860060274601
Train Epoch: 1 [1120/6000 (19%)]	Loss: 0.038768216967583
Train Epoch: 1 [1152/6000 (19%)]	Loss: 0.039159547537565
Train Epoch: 1 [1184/6000 (20%)]	Loss: 0.038528427481651
Train Epoch: 1 [1216/6000 (20%)]	Loss: 0.039043493568897
Train Epoch: 1 [1248/6000 (21%)]	Loss: 0.038629002869129
Train Epoch: 1 [1280/6000 (21%)]	Loss: 0.038930796086788
Train Epoch: 1 [1312/6000 (22%)]	Loss: 0.037965558469296
Train Epoch: 1 [1344/6000 (22%)]	Loss: 0.038816198706627
Train Epoch: 1 [1376/6000 (23%)]	Loss: 0.037802442908287
Train Epoch: 1 [1408/6000 (23%)]	Loss: 0.039175182580948
Train Epoch: 1 [1440/6000 (24%)]	Loss: 0.038710270076990
Train Epoch: 1 [1472/6000 (24%)]	Loss: 0.038385808467865
Train Epoch: 1 [1504/6000 (25%)]	Loss: 0.038630131632090
Train Epoch: 1 [1536/6000 (26%)]	Loss: 0.038291595876217
Train Epoch: 1 [1568/6000 (26%)]	Loss: 0.038497447967529
Train Epoch: 1 [1600/6000 (27%)]	Loss: 0.037885788828135
Train Epoch: 1 [1632/6000 (27%)]	Loss: 0.038541838526726
Train Epoch: 1 [1664/6000 (28%)]	Loss: 0.038069501519203
Train Epoch: 1 [1696/6000 (28%)]	Loss: 0.038520775735378
Train Epoch: 1 [1728/6000 (29%)]	Loss: 0.038348115980625
Train Epoch: 1 [1760/6000 (29%)]	Loss: 0.038655046373606
Train Epoch: 1 [1792/6000 (30%)]	Loss: 0.038718141615391
Train Epoch: 1 [1824/6000 (30%)]	Loss: 0.038416977971792
Train Epoch: 1 [1856/6000 (31%)]	Loss: 0.038400139659643
Train Epoch: 1 [1888/6000 (31%)]	Loss: 0.038309749215841
Train Epoch: 1 [1920/6000 (32%)]	Loss: 0.038490708917379
Train Epoch: 1 [1952/6000 (32%)]	Loss: 0.037472035735846
[... 76 lines cut ...]
Train Epoch: 1 [4416/6000 (73%)]	Loss: 0.090455226600170
Train Epoch: 1 [4448/6000 (74%)]	Loss: 0.096430420875549
Train Epoch: 1 [4480/6000 (74%)]	Loss: 0.103046029806137
Train Epoch: 1 [4512/6000 (75%)]	Loss: 0.111310094594955
Train Epoch: 1 [4544/6000 (76%)]	Loss: 0.119044937193394
Train Epoch: 1 [4576/6000 (76%)]	Loss: 0.127564713358879
Train Epoch: 1 [4608/6000 (77%)]	Loss: 0.137142091989517
Train Epoch: 1 [4640/6000 (77%)]	Loss: 0.148854136466980
Train Epoch: 1 [4672/6000 (78%)]	Loss: 0.162594139575958
Train Epoch: 1 [4704/6000 (78%)]	Loss: 0.175815209746361
Train Epoch: 1 [4736/6000 (79%)]	Loss: 0.192956775426865
Train Epoch: 1 [4768/6000 (79%)]	Loss: 0.212724372744560
Train Epoch: 1 [4800/6000 (80%)]	Loss: 0.231808453798294
Train Epoch: 1 [4832/6000 (80%)]	Loss: 0.253963381052017
Train Epoch: 1 [4864/6000 (81%)]	Loss: 0.280495226383209
Train Epoch: 1 [4896/6000 (81%)]	Loss: 0.310789674520493
Train Epoch: 1 [4928/6000 (82%)]	Loss: 0.341103762388229
Train Epoch: 1 [4960/6000 (82%)]	Loss: 0.380992770195007
Train Epoch: 1 [4992/6000 (83%)]	Loss: 0.425296336412430
Train Epoch: 1 [5024/6000 (84%)]	Loss: 0.474648237228394
Train Epoch: 1 [5056/6000 (84%)]	Loss: 0.532463312149048
Train Epoch: 1 [5088/6000 (85%)]	Loss: 0.587903797626495
Train Epoch: 1 [5120/6000 (85%)]	Loss: 0.655122399330139
Train Epoch: 1 [5152/6000 (86%)]	Loss: 0.737766683101654
Train Epoch: 1 [5184/6000 (86%)]	Loss: 0.834491610527039

qingchenkanlu, 2019-09-09:
@TontonTremblay  Thanks, I use another computer, and the training is work, but I don't know what has caused this.

qingchenkanlu, 2019-09-09:
I have found what the problem is , if I use only one GPU,it can converge

yusheng@yusheng:~/Deep_Object_Pose-master/scripts$ python train.py --data /home/yusheng/fat/single/004_sugar_box_16k  --outf /home/yusheng/Deep_Object_Pose-master  --loginterval 1 --batchsize 16 --lr 0.0005
start: 19:51:59.962850
load data
training data: 375 batches
load models
Training network pretrained on imagenet.
Train Epoch: 1 [0/6000 (0%)]	Loss: 0.040953978896141
Train Epoch: 1 [16/6000 (0%)]	Loss: 0.034083966165781
Train Epoch: 1 [32/6000 (1%)]	Loss: 0.027863219380379
Train Epoch: 1 [48/6000 (1%)]	Loss: 0.052056580781937
Train Epoch: 1 [64/6000 (1%)]	Loss: 0.025588771328330
Train Epoch: 1 [80/6000 (1%)]	Loss: 0.023547889664769
Train Epoch: 1 [96/6000 (2%)]	Loss: 0.027019804343581
Train Epoch: 1 [112/6000 (2%)]	Loss: 0.020110279321671
Train Epoch: 1 [128/6000 (2%)]	Loss: 0.018488207831979
Train Epoch: 1 [144/6000 (2%)]	Loss: 0.017858909443021
Train Epoch: 1 [160/6000 (3%)]	Loss: 0.016017368063331
Train Epoch: 1 [176/6000 (3%)]	Loss: 0.014744239859283
Train Epoch: 1 [192/6000 (3%)]	Loss: 0.013830945827067
Train Epoch: 1 [208/6000 (3%)]	Loss: 0.018945587798953
Train Epoch: 1 [224/6000 (4%)]	Loss: 0.013751089572906
Train Epoch: 1 [240/6000 (4%)]	Loss: 0.013395770452917
Train Epoch: 1 [256/6000 (4%)]	Loss: 0.013925779610872
Train Epoch: 1 [272/6000 (5%)]	Loss: 0.012055069208145
Train Epoch: 1 [288/6000 (5%)]	Loss: 0.012488808482885
Train Epoch: 1 [304/6000 (5%)]	Loss: 0.013699783012271
Train Epoch: 1 [320/6000 (5%)]	Loss: 0.012231723405421
Train Epoch: 1 [336/6000 (6%)]	Loss: 0.011017498560250
Train Epoch: 1 [352/6000 (6%)]	Loss: 0.011313261464238
Train Epoch: 1 [368/6000 (6%)]	Loss: 0.011298383586109
Train Epoch: 1 [384/6000 (6%)]	Loss: 0.012755277566612
Train Epoch: 1 [400/6000 (7%)]	Loss: 0.012266611680388
Train Epoch: 1 [416/6000 (7%)]	Loss: 0.011365349404514
Train Epoch: 1 [432/6000 (7%)]	Loss: 0.012122785672545
Train Epoch: 1 [448/6000 (7%)]	Loss: 0.031905010342598
Train Epoch: 1 [464/6000 (8%)]	Loss: 0.012502704747021
Train Epoch: 1 [480/6000 (8%)]	Loss: 0.012813069857657
Train Epoch: 1 [496/6000 (8%)]	Loss: 0.012356782332063
Train Epoch: 1 [512/6000 (9%)]	Loss: 0.014413649216294
Train Epoch: 1 [528/6000 (9%)]	Loss: 0.014801333658397
Train Epoch: 1 [544/6000 (9%)]	Loss: 0.013789370656013
Train Epoch: 1 [560/6000 (9%)]	Loss: 0.014443821273744
Train Epoch: 1 [576/6000 (10%)]	Loss: 0.014905676245689
Train Epoch: 1 [592/6000 (10%)]	Loss: 0.014864686876535
Train Epoch: 1 [608/6000 (10%)]	Loss: 0.014638021588326
Train Epoch: 1 [624/6000 (10%)]	Loss: 0.014538430608809
Train Epoch: 1 [640/6000 (11%)]	Loss: 0.012766372412443
Train Epoch: 1 [656/6000 (11%)]	Loss: 0.013938860967755
Train Epoch: 1 [672/6000 (11%)]	Loss: 0.014381136745214
Train Epoch: 1 [688/6000 (11%)]	Loss: 0.013379598967731
Train Epoch: 1 [704/6000 (12%)]	Loss: 0.012310686521232
Train Epoch: 1 [720/6000 (12%)]	Loss: 0.014388881623745
Train Epoch: 1 [736/6000 (12%)]	Loss: 0.012809935025871
Train Epoch: 1 [752/6000 (13%)]	Loss: 0.013151279650629
Train Epoch: 1 [768/6000 (13%)]	Loss: 0.013079827651381
Train Epoch: 1 [784/6000 (13%)]	Loss: 0.012990090996027
Train Epoch: 1 [800/6000 (13%)]	Loss: 0.012783104553819
Train Epoch: 1 [816/6000 (14%)]	Loss: 0.013732401654124
Train Epoch: 1 [832/6000 (14%)]	Loss: 0.012417307123542
Train Epoch: 1 [848/6000 (14%)]	Loss: 0.012330837547779
Train Epoch: 1 [864/6000 (14%)]	Loss: 0.011554071679711
Train Epoch: 1 [880/6000 (15%)]	Loss: 0.012453998439014
Train Epoch: 1 [896/6000 (15%)]	Loss: 0.012853015214205
Train Epoch: 1 [912/6000 (15%)]	Loss: 0.012630909681320
Train Epoch: 1 [928/6000 (15%)]	Loss: 0.012449970468879
Train Epoch: 1 [944/6000 (16%)]	Loss: 0.011639437638223
Train Epoch: 1 [960/6000 (16%)]	Loss: 0.012345942668617
Train Epoch: 1 [976/6000 (16%)]	Loss: 0.012388405390084
Train Epoch: 1 [992/6000 (17%)]	Loss: 0.012444682419300
Train Epoch: 1 [1008/6000 (17%)]	Loss: 0.011106600984931
Train Epoch: 1 [1024/6000 (17%)]	Loss: 0.010832753032446
Train Epoch: 1 [1040/6000 (17%)]	Loss: 0.010026946663857
Train Epoch: 1 [1056/6000 (18%)]	Loss: 0.009827477857471
Train Epoch: 1 [1072/6000 (18%)]	Loss: 0.010986939072609
Train Epoch: 1 [1088/6000 (18%)]	Loss: 0.009983441792428
Train Epoch: 1 [1104/6000 (18%)]	Loss: 0.009713599458337
Train Epoch: 1 [1120/6000 (19%)]	Loss: 0.009522896260023
Train Epoch: 1 [1136/6000 (19%)]	Loss: 0.010080872103572
Train Epoch: 1 [1152/6000 (19%)]	Loss: 0.010227524675429
Train Epoch: 1 [1168/6000 (19%)]	Loss: 0.010864577256143
Train Epoch: 1 [1184/6000 (20%)]	Loss: 0.010463309474289
Train Epoch: 1 [1200/6000 (20%)]	Loss: 0.009631238877773
Train Epoch: 1 [1216/6000 (20%)]	Loss: 0.009731948375702
Train Epoch: 1 [1232/6000 (21%)]	Loss: 0.009563428349793
Train Epoch: 1 [1248/6000 (21%)]	Loss: 0.010449653491378
Train Epoch: 1 [1264/6000 (21%)]	Loss: 0.009839917533100
Train Epoch: 1 [1280/6000 (21%)]	Loss: 0.009463611058891
Train Epoch: 1 [1296/6000 (22%)]	Loss: 0.010974670760334
Train Epoch: 1 [1312/6000 (22%)]	Loss: 0.011161041446030
Train Epoch: 1 [1328/6000 (22%)]	Loss: 0.010570807382464
Train Epoch: 1 [1344/6000 (22%)]	Loss: 0.010968651622534
Train Epoch: 1 [1360/6000 (23%)]	Loss: 0.009682252071798

if I use two, it can't  converge

yusheng@yusheng:~/Deep_Object_Pose-master/scripts$ python train.py --data /home/yusheng/fat/single/004_sugar_box_16k  --outf /home/yusheng/Deep_Object_Pose-master  --loginterval 1 --batchsize 16 --lr 0.0005 --gpuids 0 1
start: 19:53:56.836200
load data
training data: 375 batches
load models
Training network pretrained on imagenet.
Train Epoch: 1 [0/6000 (0%)]	Loss: 0.040255680680275
Train Epoch: 1 [16/6000 (0%)]	Loss: 1.918457865715027
Train Epoch: 1 [32/6000 (1%)]	Loss: 0.155298992991447
Train Epoch: 1 [48/6000 (1%)]	Loss: 26.744110107421875
Train Epoch: 1 [64/6000 (1%)]	Loss: nan
Train Epoch: 1 [80/6000 (1%)]	Loss: nan

do you know why?

TontonTremblay, 2019-09-09:
This is weird... The loss should be going down, nor going up.

TontonTremblay, 2019-09-09:
Which version of pytorch are you using?

qingchenkanlu, 2019-09-10:
As required, torch==0.4.0

TontonTremblay, 2019-09-10:
Can you try with the latest. You might have to change some code.

qingchenkanlu, 2019-09-10:
I use the latest code, it can converge with only one gpu, but not with two gpu

qingchenkanlu, 2019-09-10:
Sorry to disturb you, I want to ask one more question, if I want to use the model to detect real object, I need to print the Obj image and paste up ?

TontonTremblay, 2019-09-10:
The easiest way would be to buy the product in the store or order it on Amazon if you are in North America. Otherwise, printing it on a box of the exact same size as the original box is thee other way to go.

qingchenkanlu, 2019-09-10:
OK，thanks, now  I'am try to make my own dataset.

qingchenkanlu, 2019-09-10:
Excuse me, I want to know whether the background like the kitchen  has been included in the NDDS?

TontonTremblay, 2019-09-10:
No NDDS only contains the code and components needed to export annotated images. We did not include any 3d content with it.

qingchenkanlu, 2019-09-11:
It's easy to make the background?

TontonTremblay, 2019-10-08:
yeah the backgrounds are easy to add if you follow the tutorials.

qingchenkanlu, 2019-10-09:
OK,Thank you!

closed 2019-10-09 by qingchenkanlu