NVlabs/Deep_Object_Pose #77: A simple question regarding model reproducibility

trevor-taeyeop, 2019-09-16:
Hi, Tonton Tremblay
Thanks for sharing your great work!

I have a simple question regarding model reproducibility.
**- Does pretrained-weights provided in the repo reproduce the performance reported in the paper?**

When I try to evaluate DOPE with pretrained weightes provided in the repo, I found that its results shows a slightly different numbers from the paper. Specifically, the actual numbers are as below (numbers are noted only for the integer part). We used evaluation code from official DenseFusion repo and transformed 3D-model of the YCB-dataset according to the transformation matrix following here; https://github.com/NVIDIA/Dataset_Utilities/blob/532b8c76e3d7946748a10af3398438b35383f157/nvdu/config/object_settings/_ycb_original.json. 

The numbers for sugar-box and mustard-bottle are similar as in the original paper. However, numbers for the other categories are quite different from the paper. Thus, I wonder if the provided weights in this repo actually reproduce the performance reported in the original paper. Or am I missing some details in evaluation?

I would be appreciated if you reply :)

category                | original paper | our evaluation
------------------------------------------------
sugar-box              | 75 | 70
tomato-soup-can  | 76 | 67
mustard-bottle      | 81 | 80
cracker-box           | 55 | 5
potted-meat-can  | 39 | 7

TontonTremblay, 2019-09-16:
Wow thank you for reproducing our results. This is great. The weights we shared are the one that performed best in our robotics workspace. The results reported in the paper are from different weights, e.g., different epochs, that performed best for the YCB-videos which have a very `weird' noise pattern / intrinsic / compositions. Is it important for you to reproduce the results, if so please email me and I can see what we can do?

mintar, 2019-09-17:
If you find the weights that were used to produce the results from the paper, I would be very interested as well. We have recently created the [YCB-Multicam dataset](https://zenodo.org/record/2579173) and submitted a paper about it to ICRA 2020 (under review) where we evaluated DOPE on our new dataset. Here are our results (AUC) on the Xtion camera from the YCB-M dataset using the pretrained weights from the repo vs. the results on the YCB-Video dataset reported in the DOPE paper (which used different weights, as I just learned):

[image]

YCB-M is a different dataset, and the Xtion camera in YCB-M was run in normal mode, whereas the Xtion camera in YCB-Video was run in this weird "fast cropping" mode, so the results are expected to be slightly different - I believe they are plausible.

BTW, we used [obj_pose_eval](https://github.com/thodan/obj_pose_eval) (ADD metric) for evaluation.

TontonTremblay, 2019-09-23:
I had an issue with one of my computer this summer and installer 18.04 on it without realizing I deleted the weights used for the paper. I am very sorry about this.

closed 2019-09-24 by TontonTremblay