NVlabs/Deep_Object_Pose #78: Do you have change the size of obj?

qingchenkanlu, 2019-09-25:
I open the YCB obj in 3ds max and I get the size of model is smaller than the real size ,do you have change the size of .obj before make dataset?

thangt, 2019-09-25:
We didn't change the size of the .obj. How much did you detect the size changed?

closed 2019-09-25 by TontonTremblay

qingchenkanlu, 2019-09-25:
I get the model size of master chef in 3ds max is x=0.27cm y=0.286cm z=0.376cm, is it correct?

mintar, 2019-09-26:
The master chef can has the following dimensions in the original YCB model (in **meters**):

x = 0.1025
y = 0.1024
z = 0.1402

The aligned model used by DOPE (see [NVDU](https://github.com/NVIDIA/Dataset_Utilities#nvdu_ycb)) has the following dimensions (in **centimeters**):

x = 10.24
y = 14.01
z = 10.23

qingchenkanlu, 2019-09-26:
Thank you! @mintar Is x = 10.24 y = 14.01 z = 10.23 is equal to the actual object size?

qingchenkanlu, 2019-09-26:
Sorry, I want to ask you one more question @mintar ,you said [https://github.com/NVlabs/Deep_Object_Pose/issues/5#issuecomment-494708568](url),I use the meshroom to make object model,but the result is awful
[image]
[image]
I want to know how to make a good model like YCB object model?

mintar, 2019-09-26:
> Thank you! @mintar Is x = 10.24 y = 14.01 z = 10.23 is equal to the actual object size?

Yes.

> Sorry, I want to ask you one more question @mintar ,you said #5 (comment) use the meshroom to make object model,but the result is awful

You have to post-process the mesh that meshroom produces. At least, open the mesh in a mesh editing tool (like Blender or Meshlab) and cut out the table plane. Also, rescale it to fit the actual object size (in centimeters) - Meshroom doesn't know the actual size. You will probably get a hole in the mesh where the object was sitting on the table - you'll have to fill that hole, for example by recording a second mesh with Meshroom (with the object turned upside down), and then manually cutting out parts of the two meshes and gluing them together using Blender. Finally, rotate the mesh so that it's aligned with the coordinate system axes and translate it so that the origin is in the bounding box center. It's a lot of manual work unfortunately.

mintar, 2019-09-26:
Of course you can also use 3D Studio Max instead of Blender / Meshlab.

qingchenkanlu, 2019-09-26:
@mintar Thank you for your help!