NVlabs/Deep_Object_Pose #79: is it possible to train model on a custom object ?

jediofgever, 2019-09-27:


TontonTremblay, 2019-09-29:
Yes it is using NDDS. https://github.com/NVIDIA/Dataset_Synthesizer. Here are the general steps. 
1. 3d model your object, e.g., a box. You can use blender for example. 
2. Export your model to UE4 with NDDS. I recommend using FBX format 
3. Create a domain randomization with scene with your object being exported. 
4. Generate 20k images (should be enough). 
5. Use DOPE training script, it is native to NDDS exported data, for about 30 epochs. 
6. Deploy the trained weights to DOPE ROS adding weights and object dimensions. 
Let me know what you think.

jediofgever, 2019-09-30:
@TontonTremblay , thank you very much for this valuable information. I will start into gettting the 3D cad model of my object and follow the steps you listed above, perhaps it is better to keep the issue open for now , as in future steps I might need the help again.

jediofgever, 2019-10-04:
update;

I have followed all steps above, 

* I created a 3D model of object
* I used NDDS and created a dataset of the object sizing 20k images of 5 different features(depth , sematic , rgb ...)
* I trained network with following command line ; 
> python train.py --data /home/atas/Dataset_Synthesizer/Source/NVCapturedData/TestCapturer --object untitled --outf untitled --gpuids 0
* the training went on till 50 epochs with batch size 16
* however when I deploy the weights to test an image from training dataset the object cannot be detected. 

I checked the naming and made sure that the labels in json files were correct . At this point I don't really know where else to look or debug. 
I am attaching the training loss and some other files created during the training which might be helpful to understand what I did wrong. the folder that contains traning weight and other loss files;
[train_myobject](https://github.com/jediofgever/temporarycont)

The training image size is; 
1280x720(width, height)
and example image with the object;
[image]

I would much appreciate if you could have a look into this.

saratrajput, 2019-10-04:
@jediofgever Please refer to this issue: https://github.com/NVlabs/Deep_Object_Pose/issues/37. I had a similar problem.

jediofgever, 2019-10-04:
Hi @saratrajput,
Thanks for the input. I did follow the issue and essentially understand the problem.  How did you overcome the issue ?, NDDS does not provide  a built-in feature to limit random poses in a range. Can you explain steps you took

closed 2019-10-07 by jediofgever

jediofgever, 2019-10-24:
To anyone who have  problem with detection of oval objects;
 in NDDS you can modify Blueprint of the object that you want to train. Changing interval of RandomRotation  to 
 (0,0) from (-180,180) in rotational axe(z in my case) solved problem for me.