NVlabs/Deep_Object_Pose #81: UE4 Environment for creating a dataset?

TrinhTUHH, 2019-10-03:
Hi @TontonTremblay ,
I am trying to creating a similar dataset but with my own objects. In the paper, you mentioned that you used the virtual environment such as kitchen, sun temple, forest, etc. for their high fidelity. Could you please share the links to these environments? Thank you in advance!

thangt, 2019-10-08:
Hi @TrinhTUHH,

Here is how you get the SunTemple scene: https://github.com/NVlabs/Deep_Object_Pose/issues/45#issuecomment-502304956

We bought the kitchen scene from here: https://www.unrealengine.com/marketplace/en-US/slug/archviz-kitchenette

The forest scene is from here: https://www.unrealengine.com/marketplace/en-US/slug/open-world-demo-collection

TrinhTUHH, 2019-10-08:
Thank you very much Thang To!

closed 2019-10-08 by TrinhTUHH