NVlabs/Deep_Object_Pose #86: Observation angle of object

princefr, 2019-10-28:
Hi, I've been playing with [SIDOD DATASET](https://research.nvidia.com/publication/2019-06_SIDOD%3A-A-Synthetic) from Nvidia who have been generated with the same utility that you used to generated the data for DOPE.

As I understand in DOPE, your model learn to generate the 9 keypoints (8 corners + center of the projected cuboid) then you do a perspective transform to project the object into the 3D space without using other label informations.

I'm interested in trying other approaches using this dataset but I'm struggling to understand how can I get the observation angle of the object  using quaternion_xyzw information or the pose_transform.

Can you help me, or give me some hint about this ?

thangt, 2019-10-28:
@princefr Can you provide the definition of `observation angle of object`? 
From what I can find, `observation angle` is the angle between the ray from a point on the object to the observer (in this case, the capturer camera) and to the light source. Is that correct?

princefr, 2019-10-28:
Hi thangt thank you for your quick answer. for example its the theta L angle in this image

[image]

thangt, 2019-10-29:
To calculate the theta L angle, you need to calculate 2 vectors: 
- Vcam_obj: vector direction from the camera to the object
- Vobj_dir: vector show the `forward` direction of the object
From the object's quaternion, you can convert it into a rotation matrix 3x3. A rotation matrix is actually collection of 3 axis of the object: X, Y, Z. You can pick your own `forward` vector - Vobj_dir, in OpenCV coordinate convention, Z is forward.
Since the image is captured in the camera space, the camera location is [0, 0, 0], the ray from the camera to the object's center - Vcam_obj is just object's location.

To calculate theta angle, you just need to get the angle between Vobj_dir and the `right` vector which is X axis in the OpenCV convention: [1, 0, 0].

princefr, 2019-10-29:
Thank you very much.

closed 2019-10-30 by princefr

lx-r, 2023-04-14:
> Hi thangt thank you for your quick answer. for example its the theta L angle in this image
> 
> [image]
> 
> If it can help, its the alpha value provided by the kitti dataset.

Hello, Could you please tell me where you find the image?