NVlabs/Deep_Object_Pose #87: If I want to train the network,only use FAT is enough？

qingchenkanlu, 2019-10-28:
I want to train only one object like potato soup, I use 005_tomato_soup_can_16k, after I train 200 epoch, it can not detect object, the dataset is enough?

TontonTremblay, 2019-10-28:
I never really did extensive testing on any models trained on FAT, e.g., deployed on a robotics system. I prefer to use domain randomization data as it has a higher likelyhood to capture data. That being said, trained on FAT should give you some results. Which objects are you training on? Did you try to visualize the belief maps?

qingchenkanlu, 2019-10-29:
@TontonTremblay Thanks for your reply! I am training tomato soup can.  I have visualize the belief maps, but the result is bad. I add some dataset that make by myself.
[image]
[image]
[image]

I want to ask one more question:
1、How to make dataset in a series,the pictures of FAT is 000000 to 000099, but I using NDDS can only capture one or more same picture, if I want change the pose of object, I must stop and change the pose of object and start again, which cause the old img covered by new img, how to solve it?
2 、The detect speed is slow, is there any way to aaccerate the speed?

TontonTremblay, 2019-10-29:
Yeah it does not seem like it is working very well. 
1) NDDS there are tools to randomize the pose of the objects, check nvRandomOrientation or something like this. 
2) The architecture is using multiple stages, try to get the output of stage 1 instead of the 6th one. Check where DopeNetwork is defined in the forward function you can force a return there at stage 1. 
3) I do not remember exactly, there have been some discussion about that in the issues, please consult around. 
4) I do not see a problem, I did experiment with mobilenet a while ago, and it worked fine.

qingchenkanlu, 2019-10-29:
That's great! could you privide your mobilenet training code or provide the structure of the network? I have confused for a long time. Thank you!

TontonTremblay, 2019-10-29:
I am sorry, I wont be able to share the code at the moment.

qingchenkanlu, 2019-10-30:
That's OK, could you provide your network architecture？and the dataset you have used only is FAT?

closed 2019-11-03 by qingchenkanlu