NVlabs/Deep_Object_Pose #88: GPU load utilisation

cianohagan, 2019-11-12:
I am trying to train with the Falling Things dataset on one GeForce GTX 1080 GPU. I am experiencing a bottleneck in training time as the batches are getting loaded into memory on the GPU. Would you know a way to speed this process up? I have set num_workers to 0 but am not sure if this has an effect. Would it require changes to the dataloader function?

TontonTremblay, 2019-11-12:
The dataloader could be changed to use cuda enabled libraries. https://developer.nvidia.com/DALI for example. I do not have any experience with it so I wont be able more than that. I don't think you want workers set a 0. These are threads that load the data. What is your batchsize? But one GPU would take a little while to train DOPE. But with patience it should work just fine. You could also look into using a smaller front end (vgg11), and maybe train 2 stages instead of the 6 currently.

cianohagan, 2019-11-18:
There was no way around this as num_workers cannot be increased due to errors related to using Windows. I have swapped to a linux OS and no longer have this issue

closed 2019-11-18 by cianohagan