NVlabs/Deep_Object_Pose #90: Dose the detect pipeline dependent on the texture of mesh model?

sandy-ssdut, 2019-12-04:
Hi，

I run the detect pipeline in a test image， with the provided wights.

The test image contains  006_mustard_bottle, **008_pudding_box**, 010_potted_meat_can. But the detect result shows that there are 006_mustard_bottle(magenta box), **009_gelatin_box(write box)**,  010_potted_meat_can(blue box) in this image.
[image]

I wonder that, does the detect pipeline dependent on the texture of mesh model? 

If not, we can see that the texture of a mesh model stay the same in all virtual environments. Does the texture of a mesh model changes in Domain Random dataset?

TontonTremblay, 2019-12-04:
Yeah the trained models heavily use the textures on the object. I have trained version that can regress to multiple objects on one models, but these tend to be good at classes detection. 

The texture of the mesh is never changed in the training dataset.

sandy-ssdut, 2019-12-07:
Thanks！

I try to create train images for another object by NDDS plugin. But after I add the obj files into "Override Actor Meshes" of GroupActorManager_BP, I found that the rendered object do NOT have texture. How can I fix it?

sandy-ssdut, 2019-12-07:
Sorry, I used the wrong AnnotatedActor blueprint.

This problem can be fix by using the blueprint DR_AnnotatedActor_BP.

sandy-ssdut, 2019-12-08:
Sorry I want to ask one more question.

I found that there are many random components which can change the light or the movement of objects randomly.
But how to change the projective matrix of virtual camera randomly like in FAT dataset?

closed 2025-08-08 by nv-jeff