NVlabs/Deep_Object_Pose #91: Out of the memory

50618861, 2019-12-09:
I have got the customize dataset through NDDS and I'm trying to train the weight by using train.py. However, there is an issue that GPU is running out of memory. I'm using 8 K80 with 11GB memory for each GPU, image size is 512 and batch size is 8.
Does anyone know how to fix it？Thanks very much.

TontonTremblay, 2019-12-09:
I never tried to train with image size 512, I normally set it up at 400.
Batchsize of 8 is very small for the rig you have. I would think you can go
up to a least 128 batchsize. Can you share an image and one json file
please? Also please try to train it on 1 gpu with batchsize of 1 and image
size of 400 to see if this works for you?

michalvavrecka, 2020-01-14:
Batch size less than 32 results in tensor error. Standard train.py requires 17 GB GPU

TontonTremblay, 2020-01-17:
That seems high, but I will believe you. Does it work when you use less than 32?

michalvavrecka, 2020-01-17:
There was a batch size problem with old CUDA version (8.0) resulting in tensor error. With newer version 10.1 and torch 0.4.0 you can train 32 batch size (default train example with soup object) on 17 GB GPU, batch size 16 on 8 GB and batch size 8 on 5GB GPU.

TontonTremblay, 2020-01-17:
Perfect thank you.

avinashsen707, 2020-02-01:
> There was a batch size problem with old CUDA version (8.0) resulting in tensor error. With newer version 10.1 and torch 0.4.0 you can train 32 batch size (default train example with soup object) on 17 GB GPU, batch size 16 on 8 GB and batch size 8 on 5GB GPU.

```
start: 08:18:54.657828
load data
training data: 7 batches
load models
Training network pretrained on imagenet.
Traceback (most recent call last):
  File "train.py", line 1392, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1368, in _runnetwork
    epoch,batch_idx,loss.data[0]) 
IndexError: invalid index of a 0-dim tensor. Use tensor.item() to convert a 0-dim tensor to a Python number
```

I am also getting this error even though using cuda 10.1 and pytorch 0.4!
Can any one help me to gethrough..

ibrahimmahroum, 2020-03-12:
Trying to train the soup object on two GTX 1080Ti 11GB each with the following error : 
_ERROR: Unexpected bus error encountered in worker. This might be caused by insufficient shared memory (shm)._
32 Batchsize is not working, i can only train with a batch size equal or lower to 8
_working on CUDA 10.2_
Is there any solution?

closed 2020-03-12 by TontonTremblay