NVlabs/Deep_Object_Pose #93: Does the model generalise for new objects?

kevgeo, 2019-12-27:
Hi,

I am trying to detect multiple cubes in an image. The dimensions don't match to any objects in the config_pose file, so is this the main reason why the detection is nil or extremely poor?
Will we have to generate new synthetic data of cubes and train to detect them?

Also what exactly does **mesh** exactly mean? If I am correct, it is basically an object in a 3d scene. So what information does mesh exactly contain? Also to generate a mesh for a new object, can I use Unreal Engine 4 or do I need a CAD model or Meshroom as mentioned in a [previous issue](https://github.com/NVlabs/Deep_Object_Pose/issues/5) on the same topic?

TontonTremblay, 2019-12-29:
Yes if you want to detect cubes you will have to generate new training data for DOPE to find them. If you are doing cubes only, you can use the cube in UE4. The weights we shared only work with the exact same objects mentioned.

kevgeo, 2020-01-22:
@TontonTremblay Thanks for your swift reply :)  and I apologize for the delayed response. You can close the issue.

closed 2025-08-08 by nv-jeff