NVlabs/Deep_Object_Pose #94: Doubt : Regarding DOPE+NDDS UE4 Dataset Creation #DOPE #NDDS #UE4 #ataset

avinashsen707, 2020-01-10:
After those few initialising steps ,i got stucked with following problems;I followed the NDDS instructions and the respective CAD models imported via " import into level" plugin and placed these models in field of view of NVscene capture.But my objects neither randomly rotated nor changed its pose. And i also faced issues regarding giving different scenes and distractions like in the ""DOPE Dataset"".
@mintar @tmbdev @vinodgro @v4hn @mjgarland 
can anyone help me to get through these satges?

TontonTremblay, 2020-01-10:
Have you added the random position and so forth to your object? Then if yes, your object is probably set to static in UE4.

avinashsen707, 2020-01-11:
actually i am in the search of random object option in UE4 engine; can you please specify where exactly i can find it or anything related to it @TontonTremblay 👍🏼. it will be very helpful to me on proceeding. 

and yes i set the objects as "dynamically movable" in the pop box arrived while we adding an fbx file in UE4.

TontonTremblay, 2020-01-17:
I think I made a video showing how to do that in the NDDS github, have you seen it?

avinashsen707, 2020-01-29:
@TontonTremblay thankyou the problm is siolved.

closed 2020-01-29 by avinashsen707